zotil/TorBox #1: Can't access .onion site

zkrystian, 2019-04-03:
Hi,

Why I can't access to .onion site when I'm connected to TorBox wifi?

radio24, 2019-04-06:
Hello zkrystian,
unfortunately, that's not an issue for TorBox, the problem lies with the browser. As per IETF RFC 7686, "Applications that do not implement the Tor protocol should generate an error upon the use of .onion and should not perform a DNS lookup." As I know, all current browser did implement IETF RFC 7686, even Google Chrome.

radio24, 2019-05-15:
Hello zkrystian,
I reopened your question. IETF RFC 7686 still applies for Firefox and Safari -- they don't resolve .onion addresses. However, Chrome (tested with Version 74) / Chromium (Version 76) resolve .onion addresses. In other words: use Chrome / Chromium to access .onion sites.

radio24, 2019-05-15:
With pre-0.24 TorBox will offer SOCKS v5 proxy capability. Correctly configured, this will enable Firefox (and probably other browsers) to resolve .onion addresses.

radio24, 2019-05-28:
The recently released [TorBox version 0.2.4](https://github.com/radio24/TorBox/releases/tag/v.0.2.4-25052019) includes a SOCKS v5 proxy server at 192.168.42.1:9050. In addition to Chrome / Chromium, other browsers and tools with support for SOCKS v5 directly can access .onion sites. For details, see "[FAQ, Question #9](https://www.torbox.ch/?page_id=112#09)".

closed 2019-05-28 by radio24