zotil/TorBox #11: Error on Building with the TorBox install script (EXPERIMENTAL!)

openoms, 2020-01-07:
After burning the latest Raspbian Lite logged in via ssh and ran
```
cd
curl https://raw.githubusercontent.com/radio24/TorBox/master/install/run_install.sh --output run_install.zip
chmod a+x run_install
./run_install
```

Result:
```
pi@raspberrypi:~ $ cd
pi@raspberrypi:~ $ curl https://raw.githubusercontent.com/radio24/TorBox/master/install/run_install.sh --output run_install.zip

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 11577  100 11577    0     0  33556      0 --:--:-- --:--:-- --:--:-- 33459
pi@raspberrypi:~ $ chmod a+x run_install
chmod: cannot access 'run_install': No such file or directory
pi@raspberrypi:~ $ ./run_install
-bash: ./run_install: No such file or directory
pi@raspberrypi:~ $ 
```

Same when attempting with root user.

The installation has worked with the following syntax:
```
cd
wget https://raw.githubusercontent.com/radio24/TorBox/master/install/run_install.sh
chmod a+x run_install.sh
./run_install.sh
```

radio24, 2020-01-07:
Hello openoms,
thanks for your feedback. Your findings are correct. 

There are two problems. First, GitHub changes the file location from time to time. The old link is then no longer a file, but a redirection to the new location of the file. Curl in this form doesn't follow redirections. As a consequence, the file run_install.sh comprises only redirection information in the form of HTML. However wget does follow redirections and correctly downloads the file. Second, I mixed run_install.sh run_install.zip and run_install incorrectly altogether

I fixed the description on GitHub and on the Website with the correct version, as you mentioned at the end of your feedback.

Issue fixed: 9353a49e8af4301386f8782cffa674f419693876

closed 2020-01-08 by radio24

openoms, 2020-01-08:
Thank you for fixing this quick!
This is an awesome project, will continue to test.