zotil/TorBox #12: No connection: Failed to bind one of the listener ports

openoms, 2020-02-04:
My TorBox on a RPi3 B+ stopped working (it was off for a while preceding this). There is a TorBox030 WiFi available and ssh connection is working, but no internet connection due to a Tor error. Every setting is on default. 
Tried so far:
* previous SD image
* latest image
* updating firmware + Tor + TorBox
* resets

`sudo journalctl -u tor@default` shows:

```
Feb 04 08:45:23 TorBox030 systemd[1]: Starting Anonymizing overlay network for TCP...
Feb 04 08:45:23 TorBox030 tor[950]: Feb 04 08:45:23.825 [notice] Tor 0.4.2.5 running on Linux with Libevent 2.1.8-stable, OpenSSL 1.1.1d, Zlib 1.2.11, Liblzma 5.2.4, and Libzstd 1.3.8.
Feb 04 08:45:23 TorBox030 tor[950]: Feb 04 08:45:23.825 [notice] Tor can't help you if you use it wrong! Learn how to be safe at https://www.torproject.org/download/download#warning
Feb 04 08:45:23 TorBox030 tor[950]: Feb 04 08:45:23.826 [notice] Read configuration file "/usr/share/tor/tor-service-defaults-torrc".
Feb 04 08:45:23 TorBox030 tor[950]: Feb 04 08:45:23.826 [notice] Read configuration file "/etc/tor/torrc".
Feb 04 08:45:23 TorBox030 tor[950]: Feb 04 08:45:23.836 [notice] You configured a non-loopback address '192.168.42.1:9053' for DNSPort. This allows everybody on your local network to use your machine as a proxy.
Feb 04 08:45:23 TorBox030 tor[950]: Feb 04 08:45:23.836 [notice] You configured a non-loopback address '192.168.43.1:9053' for DNSPort. This allows everybody on your local network to use your machine as a proxy.
Feb 04 08:45:23 TorBox030 tor[950]: Feb 04 08:45:23.836 [notice] You configured a non-loopback address '192.168.42.1:9040' for TransPort. This allows everybody on your local network to use your machine as a prox
Feb 04 08:45:23 TorBox030 tor[950]: Feb 04 08:45:23.836 [notice] You configured a non-loopback address '192.168.43.1:9040' for TransPort. This allows everybody on your local network to use your machine as a prox
Feb 04 08:45:23 TorBox030 tor[950]: Feb 04 08:45:23.840 [notice] Not disabling debugger attaching for unprivileged users.
Feb 04 08:45:23 TorBox030 tor[950]: Configuration was valid
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.150 [notice] Tor 0.4.2.5 running on Linux with Libevent 2.1.8-stable, OpenSSL 1.1.1d, Zlib 1.2.11, Liblzma 5.2.4, and Libzstd 1.3.8.
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.150 [notice] Tor can't help you if you use it wrong! Learn how to be safe at https://www.torproject.org/download/download#warning
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.150 [notice] Read configuration file "/usr/share/tor/tor-service-defaults-torrc".
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.150 [notice] Read configuration file "/etc/tor/torrc".
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.154 [notice] You configured a non-loopback address '192.168.42.1:9053' for DNSPort. This allows everybody on your local network to use your machine as a proxy.
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.155 [notice] You configured a non-loopback address '192.168.43.1:9053' for DNSPort. This allows everybody on your local network to use your machine as a proxy.
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.155 [notice] You configured a non-loopback address '192.168.42.1:9040' for TransPort. This allows everybody on your local network to use your machine as a prox
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.155 [notice] You configured a non-loopback address '192.168.43.1:9040' for TransPort. This allows everybody on your local network to use your machine as a prox
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] You configured a non-loopback address '192.168.42.1:9050' for SocksPort. This allows everybody on your local network to use your machine as a prox
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] You configured a non-loopback address '192.168.43.1:9050' for SocksPort. This allows everybody on your local network to use your machine as a prox
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] You configured a non-loopback address '192.168.42.1:9053' for DNSPort. This allows everybody on your local network to use your machine as a proxy.
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] You configured a non-loopback address '192.168.43.1:9053' for DNSPort. This allows everybody on your local network to use your machine as a proxy.
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] You configured a non-loopback address '192.168.42.1:9040' for TransPort. This allows everybody on your local network to use your machine as a prox
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] You configured a non-loopback address '192.168.43.1:9040' for TransPort. This allows everybody on your local network to use your machine as a prox
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opening Socks listener on 192.168.42.1:9050
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opened Socks listener on 192.168.42.1:9050
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opening Socks listener on 192.168.43.1:9050
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [warn] Could not bind to 192.168.43.1:9050: Cannot assign requested address
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opening DNS listener on 192.168.42.1:9053
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opened DNS listener on 192.168.42.1:9053
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opening DNS listener on 192.168.43.1:9053
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [warn] Could not bind to 192.168.43.1:9053: Cannot assign requested address
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opening Transparent pf/netfilter listener on 192.168.42.1:9040
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opened Transparent pf/netfilter listener on 192.168.42.1:9040
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opening Transparent pf/netfilter listener on 192.168.43.1:9040
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [warn] Could not bind to 192.168.43.1:9040: Cannot assign requested address
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opening Control listener on 127.0.0.1:9051
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [notice] Opened Control listener on 127.0.0.1:9051
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.157 [notice] Closing partially-constructed Socks listener on 192.168.42.1:9050
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.157 [notice] Closing partially-constructed DNS listener on 192.168.42.1:9053
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.157 [notice] Closing partially-constructed Transparent pf/netfilter listener on 192.168.42.1:9040
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.157 [notice] Closing partially-constructed Control listener on 127.0.0.1:9051
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.157 [warn] Failed to parse/validate config: Failed to bind one of the listener ports.
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.157 [err] Reading config failed--see warnings above.
Feb 04 08:45:24 TorBox030 systemd[1]: tor@default.service: Main process exited, code=exited, status=1/FAILURE
Feb 04 08:45:24 TorBox030 systemd[1]: tor@default.service: Failed with result 'exit-code'.
Feb 04 08:45:24 TorBox030 systemd[1]: Failed to start Anonymizing overlay network for TCP.
Feb 04 08:45:24 TorBox030 systemd[1]: tor@default.service: Service RestartSec=100ms expired, scheduling restart.
Feb 04 08:45:24 TorBox030 systemd[1]: tor@default.service: Scheduled restart job, restart counter is at 5.
Feb 04 08:45:24 TorBox030 systemd[1]: Stopped Anonymizing overlay network for TCP.
Feb 04 08:45:24 TorBox030 systemd[1]: tor@default.service: Start request repeated too quickly.
Feb 04 08:45:24 TorBox030 systemd[1]: tor@default.service: Failed with result 'exit-code'.
Feb 04 08:45:24 TorBox030 systemd[1]: Failed to start Anonymizing overlay network for TCP.
```

What could be the way to fix this?

radio24, 2020-02-04:
Hello,
according to your log, Tor tries to serve clients on its TorBox030 WiFi (192.168.42.1) **and** on its ethernet interface (192.168.43.1). However, you don't have any clients on your ethernet interface, which leads to such an error. See these lines: 
```
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.156 [warn] Could not bind to 192.168.43.1:905x: Cannot assign requested address
Feb 04 08:45:24 TorBox030 tor[951]: Feb 04 08:45:24.157 [warn] Failed to parse/validate config: Failed to bind one of the listener ports.
```

Usually, choosing in the main menu one of the connection settings (entry 4-8) should solve the problem (**it may help if you reboot TorBox first**). If you don't have your internet via ethernet cable, then **make sure that no ethernet cable is attached**. If that doesn't work you can also try the following commands in the SSH shell:
```
sudo cp /etc/tor/torrc  /etc/tor/torrc.bak
sudo cp etc/tor/torrc /etc/tor/
sudo reboot
```
Afterwards chose in the main menu one of the connection settings (entry 4-8). 

Please tell me if that worked.

openoms, 2020-02-05:
Thank you, selecting the option 4 Ethernet cable (eth0) solved the problem quickly.

As I was aiming to use the box plug and play this step felt to be redundant.

The Eth cable had live internet from the beginning, so not sure if it is intentional to try to serve clients there and not just source the internet connection automatically. Might need a line readme if choosing the internet source source should be done on the first boot.

Now the Box is working great, it is a great use of my old RPi3 as a travelling privacy router!

radio24, 2020-02-05:
Thank you for your feedback. 

If the ETH cable is connected and the interface gets an IP address from the router, TorBox shouldn't use that interface for clients. However, if you change the connection settings, it automatically stores its configuration, assuming that the settings don's change all the time. Maybe it's a remnant from a previous configuration. I keep an eye on that, and if that problem occurs more often, I will search for a solution.

closed 2020-02-08 by radio24