zotil/TorBox #13: update menu

hcarrega, 2020-02-25:
just out of curiosity I tried to update through the menu ... however in the end there was an error and ended up not updating, also in the log I saw an error in the git of the tor. it's not serious, i'm really enjoying the project. I hope that in future updates everything will be 5 stars
continues in this fantastic project

radio24, 2020-02-25:
Hello hcarrega
Thanks for your feedback. Could you give me more information about the issue?
- What version of TorBox do you have? torbox-20200112-v030.gz ?
- Could you tell me something about the error?
- What did you see in the log file?

With your feedback, I can probably fix the error and provide you a file to replace.

hcarrega, 2020-02-25:
i think the problem is solved maybe some errors on repos .
thanks

radio24, 2020-02-25:
Hmm, ok, I'm leavibg topic open for now, and I will check the update functionality in the next weeks.

radio24, 2020-05-16:
I improved the reliability of the update script ([https://github.com/radio24/TorBox/blob/master/update](https://github.com/radio24/TorBox/blob/master/update)). More is coming with version 0.3.1, which is under development and will be hopefully released in the next weeks.

hcarrega, 2020-05-16:
thanks sir:)
its nice to have updates from you. torbox is awesome and runs very well

closed 2020-05-16 by radio24