zotil/TorBox #14: Tor not starting

c0deless, 2020-02-25:
hello the torbox is not working ans it's due to this error "systemctl status tor@default.serviceB
● tor@default.serviceB.service - Anonymizing overlay network for TCP (instance default.serviceB)
   Loaded: loaded (/lib/systemd/system/tor@.service; disabled; vendor preset: enabled)
   Active: inactive (dead)"

radio24, 2020-03-07:
Hello codeless
This status doesn't help to find the problem because it does only tell us what we already know: it is not working. Let's start with some basic questions:
- Which version of TorBox are you using? 
- How is your TorBox connected to the internet?
- What happens if you use another connection and the corresponding menu entries in the main menu?

liotier, 2020-05-12:
Maybe the same problem: TorBox v.0.3.0 installed today on a Raspberry Pi Model B Rev 2, with external Wi-Fi adapter. I connected through Wi-Fi, used the menu to specify connection to the Internet through the Ethernet port - it works, I can 'apt-get update' (I did not touch anything else).

/var/log/tor is empty and 'service tor@default.service status' shows inactive.

'service tor@default.service start' fails.

journalctl -xe shows constant automatic restarts, with "Control process exited, code=killed, status=11/SEGV"

radio24, 2020-05-16:
Did you do "apt-get update" by hand or did you use main menu entry 12 "Go to the update and reset sub-menu..." and then "1  Update the base system (reboot recommended!)"?

radio24, 2020-05-16:
I've been trying to get to the bottom of this problem. I'm not quite sure because of the scarce information. However, if both c0deless and liotier have done a manual apt-get update / upgrade (not via the Update and Reset sub-menu), they may have overwritten /lib/systemd/system/tor@default.service and /lib/systemd/system/tor@.service, which now bitch when loading the Tor daemon. 

I have now added a new menu item to the [update script](https://github.com/radio24/TorBox/blob/master/update) ("10 Just fixing and cleaning"), which, when executed, will do the necessary settings on both files. At least after a restart, TorBox should be running again.

liotier, 2020-05-16:
I do not remember which way I upgraded - so I'm going to reinstall and I'll report back.

closed 2020-05-25 by radio24