zotil/TorBox #15: ICMP/PING is not passing through TOR

johdn1010, 2020-03-21:
I have no 100% accurate way of verifing this, but when I ping something, the ping is very low and the same as if I ping it from my home connection. I've also figured out, that the pings are "GEO" Based on the location I'm from.. That makes me think, ping/icmp isn't "torrified" - over tor. 

I don't know if this is on purpose or no, I don't think it should be this way.
Please take a look, thank you so much!!

radio24, 2020-04-08:
Tor does not support ICMP, which is needed by ping. Tor does only support TCP. This has nothing to do with TorBox and is not a bug.

Possible solution:
Installing and using [httping](https://www.vanheusden.com/httping/)

johdn1010, 2020-04-09:
Hi yes, I guess I've figured out back then. My bad.

But shouldn't it be better to add some IPtables rules, to simply just nullroute the ping? to everytime you try to ping something, forward the traffic to 127.0.0.1 along with other stuff that isn't going through tor? Just a suggestion.

thanks

closed 2020-04-09 by johdn1010