zotil/TorBox #22: TorBox for TorBox (Whonix-GW)

TNTBOMBOM, 2020-06-25:
Hi There,

I saw your project and it has nice concept, i was wondering if you would like to use Whonix-GateWay as the distro for Torbox, for sure you will ask why? 

- Whonix-GW torified by default 
- It contains many security features (not included in the pi or debian vanilla...etc) for e.g: 
* Its own firewall which is configured to work with Tor based distro
* Hardened kernel e.g [Security-misc](https://www.whonix.org/wiki/Security-misc)
* Use of MAC like Apparmor
* sdwdate to prevent time leaks
...etc

I recommend to read Whonix Wiki:

- Clearnet: 

https://www.whonix.org/wiki/Documentation

- Onion v3: 

http://www.dds6qkxpwdeubwucdiaord2xgbbeyds25rbsgr73tbfpqpt4a6vjwsyd.onion/wiki/Documentation

So instead of making a distro yourself from scratch here is one ready, just needs to be configured to on pi.

Coincidence: Whonix old name called TorBOX :D

https://www.whonix.org/wiki/History#Brief_Whonix_.E2.84.A2_History

http://www.dds6qkxpwdeubwucdiaord2xgbbeyds25rbsgr73tbfpqpt4a6vjwsyd.onion/wiki/History#Brief_Whonix_.E2.84.A2_History

If you have further questions feel always free to ask

Thank you btw for your great work :)

radio24, 2022-01-02:
I looked into [sdwdate](https://www.whonix.org/wiki/Sdwdate), but I see some points as a no-go for the implementation into the TorBox.

Sdwdate needs a working tor installation. However, TorBox is configured that way that even all the client data traffic is strictly going through the tor network or will be blocked, part of TorBox's local traffic (for example, for the installation, maintenance and update) is not routed through the tor network. One of the main reasons for it is that some parts of the TorBox also have to function when tor is not running. This is also the case with synchronizing the system clock. A heavily out of sync system clock could prohibit the start of tor, which sdwdate needs to synchronize the clock to make tor work. 

Also, swdate is not available as a Debian package. To install the package, I have to add specific Whonix repositories or install it manually.

nyxnor, 2022-01-02:
about using whonix as the base, I don't think they have a finished arm64 image https://forums.whonix.org/t/whonix-for-arm64-raspberry-pi-rpi/1788/167

the other tools are great, threy really harden the device, but Whonix has more contributors, I think if more people could help on TorBox, there could be a way to even enable updates over tor. The only part that will never be possible to be torified is the build before tor is installed, but the other steps can be done, giving a fallback option to not route through tor if it is broke or something like it.

radio24, 2022-01-02:
To be clear: I don't see a way to use Whonix as a base system for the TorBox, because it would mess around the configuration files of Whonix.

The question is if it is really necessary to torrify the update and maintenance traffic doing local on the TorBox. We already gave the user the possibility to deny clearnet access to tor-related resources (for example, updating tor, the torbox menu through tor). For me, the negative security implications are not clear if someone updates, for example, the Debian base system through clearnet. The decision not to torrify a Debian base system update is also based on the lower speed and reliability of the tor network leading to broken installations. Also, torrifying the local network traffic would mean TorBox wouldn't handle captive portals.

Nevertheless, probably, it would be feasible to integrate a complete torrifying / untorrifyng option in a sub-menu with additional warnings. Even it is on my to-do list, it is not on my priority schedule.

adrelanos, 2022-01-05:
I am a developer of @Whonix.

> A heavily out of sync system clock could prohibit the start of tor, which sdwdate needs to synchronize the clock to make tor work.

Tor starts independently of system clock.
Tor connection fails if the system clock is too slow or too fast.

Nowadays sdwdate can fix the system clock even if the the system clock is extremely slow for example such as year 2011 but it doesn't matter could be as slow as 1984 or any time earlier. sdwdate uses onion-time-pre-script which uses anondate-set which is capable to fix this.

Users with too slow clocks that happens sometimes and good that sdwdate can fix the time nowadays. Users with much too fast clocks, we don't have such reports but sdwdate couldn't fix that. We don't have a concept currently how to securely move the clock backwards, maybe in the future.

> Also, swdate is not available as a Debian package.

Not available from packages.debian.org but available as a Debian package.

> about using whonix as the base, I don't think they have a finished arm64 image https://forums.whonix.org/t/whonix-for-arm64-raspberry-pi-rpi/1788/167

Some people are building arm64 images from source code but yeah currently no official arm64 release, maintainer required.

> To be clear: I don't see a way to use Whonix as a base system for the TorBox, because it would mess around the configuration files of Whonix.

I don't understand what would be a problem or what Whonix is lacking for making this possible.

radio24, 2022-01-08:
Hello adrelanos
Thank you for your feedback. Last week, I also had a good (mail) feedback from @nyxnor about the idea to use Whonix as a base for TorBox. He also sent me a very interesting link to the [Whonix VM Build Documentation](https://www.whonix.org/wiki/Dev/Build_Documentation/VM).

I agree that basing it on Whonix could be worthwhile regarding security and anonymity. Maybe my comment was not well choosed in my last posting. I don't think that Whonix is lacking something, I'm more a little bit anxious, if the changes a TorBox brings to the configuration of the operating system would not cause problems. This changes can be seen on this [website](https://www.torbox.ch/?page_id=205) or in the installation scripts ([Rapberry Pi OS](https://github.com/radio24/TorBox/blob/master/install/run_install.sh), [Debian](https://github.com/radio24/TorBox/blob/master/install/run_install_on_debian.sh), [Ubuntu](https://github.com/radio24/TorBox/blob/master/install/run_install_on_ubuntu.sh)). Also, my time to deal with TorBox is very limited. I care about TorBox because I use it in my daily work, which is not IT. There is a cost of opportunity, which means, if I invest my time to port TorBox to Whonix, I cannot spend my time to improve the Onion Service support, introduced in TorBox v.0.5.0 (which should be released this weekend).

Another big issue for me is the support of Captive Portal, which I need for my daily work with the TorBox. This is also the reason why TorBox is not completely torified, because that would break the Captive Portal support. How does Whonix deal with Captive Portals?

adrelanos, 2022-01-12:
Alright, sure. Totally understandable.

> How does Whonix deal with Captive Portals?

The short answer is "there is a user `clearnet` on Whonix-Gateway."

Long answer:
https://www.whonix.org/wiki/Logging_in_to_captive_portals

adrelanos, 2022-01-12:
https://github.com/radio24/TorBox/issues/142

TNTBOMBOM, 2022-01-18:
> For me, the negative security implications are not clear if someone updates, for example, the Debian base system through clearnet. The decision not to torrify a Debian base system update is also based on the lower speed and reliability of the tor network leading to broken installations.

https://blog.packagecloud.io/eng/2018/02/21/attacks-against-secure-apt-repositories/

closed 2022-07-13 by radio24

adrelanos, 2022-07-13:
How was this completed?