zotil/TorBox #26: Unable to ssh into the TorBox

MirzaAreebBaig, 2020-09-20:
Hey, this project looks amazing for having tor over the network 

I have downloaded the image from https://www.torbox.ch
& i have burned it on the sd card & i have a raspberry pi 3 model b+
everything is working fine i got a network called "torbox032" & its password is "CHANGE-IT"
& when i open https://check.torproject.org/ its a success
but i'm unable to ssh into the TorBox i use ubuntu machine i tried with terminal & putty by using torbox as username & pi ip address
but it does not connect
it says:
ssh: connect to host 192.168.2.5 port 22: Connection timed out

i tried using "pi" "torbox" but nothing works
even i added a blank "ssh" file into the sd card but still it doesn't connect
the raspberry pi is connected with an Ethernet cable from my WiFi router
so i tried ssh into it by being in both networks 1) on my wifi 2) on the torbox
still nothing works
thanks in advance for your time & help :)

radio24, 2020-09-20:
Hello MirzaAreebBaig
Thank you for your feedback. 
You cannot connect the TorBox because you are using the wrong IP (192.168.2.5).
`To establish a connection between the SSH-client and TorBox, use 192.168.42.1 (username: torbox / password: CHANGE-IT); with an Ethernet connection, use 192.168.43.1 (username: torbox / password: CHANGE-IT).`
In your case, the easiest way would be to connect your client to the new WiFi “TorBox032” (password: CHANGE-IT), and then from your client to access 192.168.42.1 (username: torbox / password: CHANGE-IT) with ssh.

MirzaAreebBaig, 2020-09-20:
Hey thanks a lot for the quick reply :) 

I got connected to the TorBox using 192.168.42.1 great :) 

Thanks for your help

MirzaAreebBaig, 2020-09-20:
I just got one question

Can we use Pi-hole into the same raspberry pi?
if not can we use pi-hole dns in TorBox to get it work?

Thank You

radio24, 2020-09-20:
Integrating pi-hole may prove difficult because pi-hole and TorBox are probably unscrewing on the same network tools and configurations. Furthermore, such combinations can lead to security risks – the more complex and multi-modular a system is, the higher the risk.

There also other reasons why we don't want to integrate pi-hole into our TorBox project (at least at the moment) - see our explanation [here](https://www.torbox.ch/?page_id=112#could-you-incorporate-pi-hole-to-remove-ads).

MirzaAreebBaig, 2020-09-20:
Alright got it thanks :)

closed 2020-09-24 by radio24