zotil/TorBox #27: Option to block all HTTP plain text traffic

JavanXD, 2020-10-04:
While it is possible to install an add-on like HTTPSEverywhere in the Browser, it is not available for all devices such as smartphone browsers.

Since the TorBox routes all traffic of a device through the Tor network and not only of the browser you can easily be identified by the numerous unencrypted requests at the ExitNode.

Among them are plain text requests to telemetry servers and time synchronization services. A special danger are the attacks with sslstrip, which increased again in 2020: https://blog.torproject.org/bad-exit-relays-may-june-2020

The Tor Browser Project itself has identified this issue and blocking plain text HTTP requests is hopefully getting implemented as an option in the Tor Browser soon (there is an open issue). 

In the TorBox this would be easy to implement as well. Therefore I would highly recommend this feature.

radio24, 2020-10-04:
Hello Javan
You are right that we should implement blocking plain text HTTP requests or block bad exit relays. I will add that to the to-do list. However, because of job related reasons, the next major TorBox version will not be released before January/February 2021.

radio24, 2021-03-20:
Working on it....
Most probably, with TorBox 0.4.0, all HTTP plain text traffic will be blocked by default. However, there will be an option to disable it.

radio24, 2021-04-12:
It is part of TorBox v.0.4.0, which was [released yesterday](https://github.com/radio24/TorBox/releases/tag/v.0.4.0-10.04.2021).

closed 2021-04-12 by radio24