zotil/TorBox #28: Option to select preferred ExitNode Country

JavanXD, 2020-10-04:
There is already the possibility to edit the `torrc` file manually (in the menu), but an menu for the following options would be very helpful: 
- Allow only preselected, very trustworthy, ExitNotes (e.g. those from CCC)
- Regional settings for ExitNotes 

Examples from other competitors:
[image]
[image]

radio24, 2021-03-21:
Hello JavanXD 
Thank you for your suggestion. However, let me explain to you some points. TorBox is not in competition with other "anonymisation boxes". If some of them offer you the possibility to alter Tor's route selection, it doesn't mean that this is a good idea from the security and anonymisation perspective. If someone knows about the involved risks, he can probably edit the two lines in torrc by himself. I consider it negligent in making this possibility available to even the most inexperienced user through a simple menu (and apparently without warning). **We have therefore decided not to include such an option in the TorBox menu.**

Using a specific Exit Node can not only slow down your speed, but it can also compromise your security and anonymity (for example, this approach increases the correlation attack vulnerability) since all your traffic is always going through a restricted list of nodes. [The TOR FAQ states the following](https://2019.www.torproject.org/docs/faq.html.en#ChooseEntryExit): "You get the best security that Tor can provide when you leave the route selection to Tor; overriding the entry/exit nodes can mess up your anonymity in ways we don't understand."

JavanXD, 2021-03-22:
Reasonable explanation, thank you.

closed 2021-03-22 by JavanXD