zotil/TorBox #30: Raspberry Pi 3 - repo

TemhAAhmeT, 2020-12-25:
deb https://deb.torproject.org/torproject.org buster main

torproject doesnt support armhf

radio24, 2020-12-27:
Hello TemhAAhmeT
Thank you for that feedback. Unfortunately, because of my workload, I cannot fix this issue right away. I expect to find time to provide an update at the beginning of March.

radio24, 2021-01-04:
Hello TemhAAhmeT
I looked into it. Yes, this error message is appearing if in the Update and Reset submenu entry 1 is used. For that reason, we have in the same submenu entry 3, which installed the newest tor version from the source package. Armhf support is not necessary, because the binaries will be compiled on the Raspberry Pi.
The key added during the update should work just fine. 
To be sure, I will try it tomorrow again.

radio24, 2021-01-06:
Hello TemhAAhmeT
Today, on the base of TorBox v. 0.3.2, first, I updated the base system (in the Update and Reset submenu entry 1). Afterwards, I updated Tor to version 0.4.4.6 without any problem (in the Update and Reset submenu entry 3).

radio24, 2021-01-10:
For more information see also here: [https://www.torbox.ch/?p=1949](https://www.torbox.ch/?p=1949)

closed 2021-01-10 by radio24