zotil/TorBox #32: Problems with EXPERIMENTAL: TorBox on Ubuntu Server 20.04 LTS

UKWRadio, 2021-02-16:
I have the following Error when I want to Install TorBox on Ubuntu:

ubuntu@ubuntu:~$ sudo python2 get-pip.py
Traceback (most recent call last):
  File "get-pip.py", line 24244, in <module>
    main()
  File "get-pip.py", line 199, in main
    bootstrap(tmpdir=tmpdir)
  File "get-pip.py", line 82, in bootstrap
    from pip._internal.cli.main import main as pip_entry_point
  File "/tmp/tmpP45yKJ/pip.zip/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
ubuntu@ubuntu:~$ sudo python2 get-pip.py
Traceback (most recent call last):
  File "get-pip.py", line 24244, in <module>
    main()
  File "get-pip.py", line 199, in main
    bootstrap(tmpdir=tmpdir)
  File "get-pip.py", line 82, in bootstrap
    from pip._internal.cli.main import main as pip_entry_point
  File "/tmp/tmpXUuPrC/pip.zip/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax

radio24, 2021-02-16:
Hello
With the next version of TorBox, which is expected to be released in March/April, we no longer need Python 2 and the bug should be gone. We will test the Ubuntu version during March/April. In the mean time, you could try the torbox-wireless-manager branche.

UKWRadio, 2021-02-16:
Unfortunately, that doesn't work either. I took the current server version of Ubuntu. Are there any other ways I can successfully install this on Ubuntu Server? Unfortunately I don't have a Raspi at hand, just an old notebook :-(

radio24, 2021-02-20:
Hello again
Let's try this modified installer script for Ubuntu. It installs the torbox-wireless-manager branch on which we are currently working and which will be the base for the next TorBox 0.4.0 release. In this branch, wicd is replaced by our own TorBox Wireless Manager (TWM), which is working with Python 3. Consequently, we completely removed Python 2, and the error above should not appear again.

However, we couldn't test the install script and TWM under Ubuntu yet. For that reason, we are not sure if it works without any problems. For that reason, we would be thankful for any feedbacks.

`cd
wget https://github.com/radio24/TorBox/files/6015277/run_install_on_ubuntu.sh.zip
unzip run_install_on_ubuntu.sh.zip
chmod a+x run_install_on_ubuntu.sh
./run_install_on_ubuntu.sh`

[run_install_on_ubuntu.sh.zip](https://github.com/radio24/TorBox/files/6015420/run_install_on_ubuntu.sh.zip)

radio24, 2021-04-12:
With the release of TorBox 0.4.0, the problem is solved.

closed 2021-04-12 by radio24