zotil/TorBox #33: Feature request: enable/disable tor preserving forwarding functionality

bl4ckh0l3z, 2021-02-28:
Is there any chance to have an additional feature consist of "enable/disable tor daemon" menu entry and preserving network forwarding functionalities?

It could be interesting for more advanced privacy preserving use such as leveraging on torbox for network traffic analysis via tcpdump and snort.

Thanks in advance.

radio24, 2021-03-06:
**enable/disable tor daemon**: No. Honestly, if someone wants to use TorBox in a more advanced way, he surely knows how to enable/disable the Tor daemon on the command prompt.

**preserving network forwarding functionalities**: I don't know what you exactly mean by that. Do you mean that it should be possible to forward data through TorBox without using Tor? Currently, this is temporary the case if you choose the menu entries 4-9 and then the "Captive Portal" option (see [here](https://www.torbox.ch/?page_id=775)). Of course, once the captive portal has been overcome, the entire data stream is routed through the Tor network again. We have no plan to implement a way that the data stream could be forwarded through TorBox without using Tor because, from our viewpoint, that would be a vulnerability.

bl4ckh0l3z, 2021-03-06:
Ok thanks for your feedback. I have implemented this feature on my way (disable tor daemon, change iptables rules accordingly, and more). Cheers

radio24, 2021-03-06:
Good to hear... that's the beauty of open source!
Do you want to share with us the changes by adding the respective code?

bl4ckh0l3z, 2021-05-06:
> 
> 
> Good to hear... that's the beauty of open source!
> Do you want to share with us the changes by adding the respective code?

Here I am, please find the pull request:
https://github.com/radio24/TorBox/pull/38

radio24, 2021-05-06:
Thanks, we will look into it again for the next release. Probably, we will add it in a sub-menu, but most likely not as the first point in the main menu.

bl4ckh0l3z, 2021-05-06:
I totally agree, sub-menu is better for sure...it was very easy for me to add it as the first item. Cheers and thanks for this valuable project!

radio24, 2023-07-11:
In [TorBox V.0.5.3](https://github.com/radio24/TorBox/releases/tag/v.0.5.3-09.07.2023) integrated (Danger Zone sub menu)

closed 2023-07-11 by radio24

bl4ckh0l3z, 2023-07-11:
Really appreciate, thanks