zotil/TorBox #36: problem update tor from repo armhf ???

sajjads24, 2021-04-12:
hi
on torbox v.40 error tor repository and EOL tor repo for armhf
please change tor repo from armhf to aarch64 for update tor.

https://github.com/Stadicus/RaspiBolt/issues/656

radio24, 2021-04-12:
Hello sajjads24
Did you have any problem installing / re-installing / updating tor on TorBox 0.4.0 using one of our installer scripts or using the update and reset sub-menu entry 4? I this was the case, please give us more details about the used hardware, the operating system, when the error appeared and, if possible, what the error looked like.

Because we don't fetch the binaries but the source code and compile the binaries on the TorBox itself, we shouldn't have any problems. At least, we installed tor probably a dozen time during the last week and never had any issues.

sajjads24, 2021-04-12:
> Hello sajjads24
> Did you have any problem installing / re-installing / updating tor on TorBox 0.4.0 using one of our installer scripts or using the update and reset sub-menu entry 4? I this was the [...]

i am install torbox on raspberry pi 3 
so show this error on update :

`torbox@TorBox040:~/torbox $ sudo apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease                                
Ign:3 https://download.webmin.com/download/repository sarge InRelease                       
Hit:4 https://download.webmin.com/download/repository sarge Release                         
Err:6 https://deb.torproject.org/torproject.org buster InRelease                            
  Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out
Reading package lists... Done                      
Building dependency tree       
Reading state information... Done
All packages are up to date.
W: Failed to fetch https://deb.torproject.org/torproject.org/dists/buster/InRelease  Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
torbox@TorBox040:~/torbox $ 
`

sajjads24, 2021-04-12:
can i use aarch64 repo :

Try this:
sudo nano /etc/apt/sources.list

add this:
deb [arch=amd64] https://deb.torproject.org/torproject.org buster main
deb-src [arch=amd64] https://deb.torproject.org/torproject.org buster main

And do that prior installing TOR. And then install it.

per: https://support.torproject.org/apt/tor-deb-repo/

radio24, 2021-04-12:
sudo apt-get -y update
sudo apt-get -y dist-upgrade

This will only update the base system, not Tor. Of course there will be error messages displayed because TorBox is also running under Debian and Ubuntu, so we added the binary and the source repository. For example under Debian 10/11, arch is arm64 and there, we fetch the binaries.

As I said, we compile Tor from the sources. Try update and reset sub-menu entry 4 or read the code in our script: https://github.com/radio24/TorBox/blob/master/update

There is a serious risk, if you start to manually screw around that the system will break. Currently, the last version of the tor 0.4.5.x branche is installed.

radio24, 2021-04-13:
Ok, I discussed your point with others. We will remove "deb https://deb.torproject.org/torproject.org buster main" in Systems, which compile tor from the sources. With it, the error messages will disappear.

Anyway, we are planing to change the way how Tor is installed in the system beginning with the next release. The idea is, to fetch the source file from GitHub and that the user can chose, which branch he want's to have installed.

sajjads24, 2021-04-13:
i am install Raspberry Pi OS Lite from :
https://www.raspberrypi.org/software/operating-systems/#raspberry-pi-os-32-bit
so use this script for install torbox :
`cd
wget https://raw.githubusercontent.com/radio24/TorBox/master/install/run_install.sh
chmod a+x run_install.sh  
./run_install.sh `

but show error and can not add tor repo !?

please fix tor repo in script .

thanks

radio24, 2021-04-13:
I will check that again and give you an answer.
Please, after finishing the installation and restarting the Raspberry Pi, check with "tor -v" if something is installed or not. Should that be the case, please tell me the installed version. Thanks

radio24, 2021-04-13:
I did exactly the same: 

```
cd
wget https://raw.githubusercontent.com/radio24/TorBox/master/install/run_install.sh
chmod a+x run_install.sh  
./run_install.sh 
```

It compiles tor without any problem and installs on this way version 0.4.5.7.

[image]

closed 2021-04-14 by radio24

nyxnor, 2021-06-26:
Just to mention, if you ever change to 64 bit, here it is
* [image](https://downloads.raspberrypi.org/raspios_arm64/images/)
* [signing key](https://www.raspberrypi.org/raspberrypi_downloads.gpg.key)