zotil/TorBox #37: how fix error 403 ?!

sajjads24, 2021-04-28:
hi 
how fix error 403 on update ?
`root@TorBox040:~# sudo apt -y update && sudo apt -y full-upgrade
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease                             
Hit:2 http://archive.raspberrypi.org/debian buster InRelease                                
Err:3 https://deb.torproject.org/torproject.org buster InRelease         
  Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out
Reading package lists... Done                      
Building dependency tree       
Reading state information... Done
All packages are up to date.
W: Failed to fetch https://deb.torproject.org/torproject.org/dists/buster/InRelease  Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@TorBox040:~# `

sajjads24, 2021-04-28:
i am can not update tor with torbox menu ?:
` Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:03:35 --:--:--     0curl: (7) Failed to connect to deb.torproject.org port 443: Connection timed out
gpg: no valid OpenPGP data found.
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease             
Err:3 https://deb.torproject.org/torproject.org buster InRelease         
  Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out
Reading package lists... Done                      
W: Failed to fetch https://deb.torproject.org/torproject.org/dists/buster/InRelease  Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out Could not connect to deb.torproject.org:443 (10.10.34.35), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
NOTICE: 'tor' packaging is maintained in the 'Git' version control system at:
https://git.torproject.org/debian/tor.git
Please use:
git clone https://git.torproject.org/debian/tor.git
to retrieve the latest (possibly unreleased) updates to the package.
Need to get 7,871 kB of source archives.
`
after 3 or 5 minuts show wrong massage !!!?
[image]

radio24, 2021-04-29:
**We have here three issues:**
1. The "could not connect" error when trying to update **the binaries of Tor**.
2. The "failed to connect" error when curl is trying to download https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc in the process to **update Tor from its sources**.
3. The NOTICE that Tor packaging is maintained on Git.

**To issue 1**
- We already covered issue 1 here: https://github.com/radio24/TorBox/issues/36
- You can ignore the error because on Raspberry Pi OS, **Tor is installed from its sources** and not from the binary packages.
- `sudo apt -y update && sudo apt -y full-upgrade` will update your base system, **but not Tor**, because the installed version of Tor is compiled from its sources. To update Tor, you have to recompile new sources, which is done by [entry 4 in the update and reset sub-menu](https://www.torbox.ch/?page_id=1410).
- Our fix ([see here under LOOK&FEEL](https://www.torbox.ch/?p=2027#look)) should remove this error message. To put it into effect, please use [entry 5 in the update and reset sub-menu](https://www.torbox.ch/?page_id=1410).
- If this doesn't help, you can also use this command on the command line: `sudo sed -i '/^deb https:\/\/deb.torproject.org\/torproject.org/d' /etc/apt/sources.list`

**To issue 2**
- We just tried to update Tor with [entry 4 in the update and reset sub-menu](https://www.torbox.ch/?page_id=1410), and it worked without any problem.
- Are you sure that your TorBox had access to the Internet? Did TorBox use the nameservers 1.1.1.1, 1.0.0.1, 8.8.8.8 and 8.8.8.4.4 (should be visible after starting entry 4 in the update and reset sub-menu; this is to circumvent some low-level censorship measures)?
- Some times https://deb.torproject.org is offline or overloaded. Did you try this some hours later again?
- By the way: currently, the newest Tor version of the 0.4.5.x branch is already installed in your system (0.4.5.7). There is no more recent version, which is not in the alpha stage.

**To issue 3**
- We didn't see the NOTICE when we did our test, but we know that Tor is using Git.
- However, the newest Tor version of the 0.4.5.x branch [on Git ](https://gitweb.torproject.org/tor.git/)is 0.4.5.7, which is already installed into TorBox v.0.4.0 (0.4.6.x and 0.4.7.x are in the alpha stage, and we will not install it into TorBox, yet).
- We are considering using https://github.com/torproject/tor/releases for updating tor and give the user the possibility to switch between different branches in the upcoming TorBox v.0.4.1 (the release date is not set yet). For that reason, we keep this thread open until the next TorBox release.

radio24, 2021-06-07:
With the [Commit #438](https://github.com/radio24/TorBox/commit/9aae2daf22b6240010127de0972bec7431e21578), TorBox supports updating and changing between various tor versions, which are pulled from the [unofficial Tor GitHub repository](https://github.com/torproject/tor).

In an already running TorBox System (for example, with the version 0.4.0), please update the TorBox main menu with the entry 5 in the [update and reset sub-menu](https://www.torbox.ch/?page_id=1410), then chose entry 4 in the same sub-menu (the entries title should now be "Change/update the version of Tor"), and then "EXPERT". Subsequently, a list with available tor versions will be displayed - chose the one you want to have installed.

With "DEFAULT", the latest stable version of TorBox will be installed without any further user action needed.

If you install the TorBox with our [installation script](https://www.torbox.ch/?page_id=1168#script) on a fresh Raspberry Pi OS Lite image, you can use the following command to select the tor version: `./run_install.sh --select-tor`

closed 2021-06-07 by radio24