zotil/TorBox #39: torbox-20210410-v040 has undeletable files in the slack space of the second partition

rsaxvc, 2021-05-16:
Running testdisk on torbox-20210410-v040, selecting Intel partition table type, advanced utilities, second partition, list, /etc(for example), shows a number of leftover files that were deleted, but are still present in the slack space of the image. Fully removing these before packaging the image could make it a little smaller and reduce the risk of unintended pre-release artifacts hopping a ride in the slack space.

Mostly these files seem to be duplicates of config files, but I haven't searched too hard.

radio24, 2021-05-16:
Thank you for your suggestion. I have to look into this in more detail because these files on the "second partition" are most likely related to "pishrink" and needed to expand the image file on the storage medium (usually an SDCard) of the target system.

In an expanded and installed TorBox System, there are only these two partition (/boot and /):
/dev/mmcblk0p1            8192    532479     524288  256M  c  W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 31116287 30583808 14.6G 83 Linux

rsaxvc, 2021-05-16:
I mean / is the second partition, but /etc is the first directory in it with leftover files.

I'm also surprised pishrink didn't strip those out.

radio24, 2021-05-17:
Do you have any suggestions, how I can easily get rid of all these leftover files?

rsaxvc, 2021-05-17:
zerofree will fill the slack space(by default with zero). When I ran it on the torbox image, it reported it would fill 83494 disk blocks, or around ~340MB, which will compress more easily once zeroed as well.

radio24, 2021-05-17:
Thanks! I will definitely give it a try with the next release.

radio24, 2021-06-13:
**Update:** I just used zerofree on the TorBox v.0.4.1 image, which will be released in the following hours. The result is impressive: the compressed image is almost 220 Mb smaller than the one from TorBox v.0.4.0. Thank you for this suggestion!

closed 2021-07-11 by radio24