zotil/TorBox #40: torbox-20210410-v040 ships with duplicate host-keys

rsaxvc, 2021-05-16:
torbox-20210410-v040 distributes ssh hostkeys in the image. I only booted it in QEMU, so perhaps regeneration is triggered somehow on device, but you might check if everyone, on their first attempted SSH login, sees the same ssh host key fingerprint(though also, ed25519, SSH, and DSA keys are the same way):

$ ssh-keygen -lf etc/ssh/ssh_host_ecdsa_key.pub 
256 SHA256:pkPEvXktCYu9/KMTFipRxCr7K8AoFaUpULrsSZ6Lm78 root@raspberrypi (ECDSA)

radio24, 2021-05-16:
Thank you for your suggestion. I have to look into this in more detail. Looking into the /etc/ssh/ssh_host_ecdsa_key.pub in my working TorBox v.0.4.0 system shows me something completely different. However, I have to compare this with other installations, which I can't right now.

rsaxvc, 2021-05-17:
If it's helpful, here's the content of my /etc/ssh/ssh_host_ecdsa_key.pub:

`ecdsa-sha2-nistp256 AAAAE2VjZHNhLXNoYTItbmlzdHAyNTYAAAAIbmlzdHAyNTYAAABBBDIb9yCiulejBEYWNna8l4ylWP+vCAbEO9IAdsikD3nXRPw0CWhkuuqIfSEFUcAyDH01VuV5kiCuxw3YE/Hczi4= root@raspberrypi
`

The above ssh-keygen output is just the fingerprint a user would see the first time they SSH to torbox. If two machines have the same fingerprint or /etc/ssh/ssh_host_* files, it's not great ( although likely mitigated somewhat until the SSH service is exposed to the internet).

radio24, 2021-06-09:
Finally, I looked into this in more detail again. It was a little bit tricky because we have two situations, how TorBox can be installed: with the installation script and from the image file.

If TorBox is installed with the installation script, the OS has already unique SSH server keys from starting up the OS the first time. Consequently, every TorBox installed with the installation script will have its unique SSH server keys. Because I used that method, I couldn't see the issue, which you mentioned.

However, if TorBox is installed from the image file, then everyone with that image file will have the same SSH server keys, representing the issue mentioned by you. So even this is not a catastrophic situation, the best practice is to have one key per machine. That's the most secure approach.

With TorBox v.0.4.1 coming out soon, with the first time starting the TorBox menu, it will detect the first start after flashing the image and change/regenerate all SSH server keys on the machine.

Thank you very much for mentioning this issue.

rsaxvc, 2021-06-10:
Thank you for looking into it.

Shared host keys allows both MITM fingerprinting, but only with existing network access.

You might look at how the raspberry pi OS handles it as well.

closed 2021-06-10 by radio24