zotil/TorBox #41: HTTP plain text traffic by block - question

nyxnor, 2021-05-30:
Is it efficient for calls made from the terminal such as to wget files?
and
Does TorBox routes through socks 5 from the terminal to the internet (not to lan) if I change to `DNSPort 127.0.0.1:9053` instead of the static IP address? (edit `iptables.ipv4.nat` too right)? (Just to be sure) https://github.com/radio24/TorBox/blob/3064630f641bf87f491fff98d5f29eb8d39db2c4/etc/tor/torrc#L13

radio24, 2021-05-30:
Hello nyxnor
I'm not sure if I fully understand your questions. You can use wget in the terminal, but **its data traffic will not be routed through tor**. If you don't mind that it doesn't go through tor, it will be more efficient than running from a client that routes all data through tor. The question is if you want efficiency or security.

TorBox only routes the traffic from its clients through the tor network. Traffic from the terminal (for example, for maintenance and update) is not routed through tor. You shouldn't change `DNSPort 192.168.42.1:9053` in torrc, because that would break the DNS resolution for the WLAN clients. You could add `DNSPort 127.0.0.1:9053` as an additional entry to the torrc to route the local DNS request from the terminal through tor. This will only work if iptables redirects the local DNS requests on port 53 to port 9053. However, **this will not change the fact that local data traffic from the terminal would not be routed through tor**. To accomplish this, you also have to add `TransPort 127.0.0.1:9040` and/or `SocksPort 127.0.0.1:9050` as well as to redirect the entire data traffic.

We don't route local traffic from the terminal through tor because if tor is not working for whatever reason, you will not be able to easily fix it (for example, with an update). Travelling around, so far, we didn't regret this choice.

**Important: Data traffic from the connected clients, which has to be protected, is completely routed through tor (including DNS requests). Additionally, TorBox is configured that any direct DNS requests from clients are blocked.**

nyxnor, 2021-05-30:
Hey contributor, you did a great job explaining with good examples. You did answer my questions and yes, I would like to route through Tor the data from localhost/terminal (thanks for giving insctructions on how to do so).

I asked this because of text/prompt-text saying `Local network data from the command prompt is NOT routed through the Tor network!`, so by Local Network I understood as LAN traffic such as `scp` or `ssh` from lan. So in fact, is not just Local Network data, it is all the server/TorBox data transmitted through and from the internet requested by localhost that is not routed through Tor, but clients do request from the TorBox that resolves the DNS query and acquires it within the Tor network.

Thanks for clarifying, as you did answer my questions, I will close this issue. If I encounter problems redirecting the entire data traffic with the given instructions, I will reopen this. Once again, thanks :100: .

closed 2021-05-30 by nyxnor