zotil/TorBox #42: how to share tor network via eth1

connected201, 2021-05-31:
hi, i have 2 eth. (1 onboard, 1 USB) i want to connect raspberry via eth0 to receive internet, and share via eth1 (USB network card) how i can configure to obtain this setup? best regards

radio24, 2021-05-31:
Are you speaking about TorBox or about the Raspberry Pi in general?

connected201, 2021-06-01:
about TorBox, i have second ethernet adapter
[image]
for first eth0 i want to receive internet, and for second eth1 i want to share internet over TOR network

radio24, 2021-06-01:
This is easy. 
Shutdown TorBox - connect the Internet router to eth0 - connect your USB ethernet adapter (eth1) and connect your client with it - start TorBox - chose entry 4 in the TorBox main menu. That's it!

**IMPORTANT**
To access the TorBox from a such client, you have to use 192.168.**43**.1. The client will get a 192.168.43.x address.

For more information see here: 
- [Documentation - Main Menu ](https://www.torbox.ch/?page_id=775)
- [FAQ - My TorBox is connected with the Internet via ethernet.  I’m using onboard WiFi for my client devices. How can I additionally connect a cable client?](https://www.torbox.ch/?page_id=112#my-torbox-is-connected-with-the-internet-via-ethernet-im-using-onboard-wifi-for-my-client-devices-how-can-i-additionally-connect-a-cable-client)

closed 2021-06-01 by connected201