zotil/TorBox #43: Feature request - Add possibility to use OTG as network card

connected201, 2021-06-01:
Hi, maybe you can add option to use integrated OTG on Raspbery 4 as network card?
it's very easy to use eth0 for internet, and OTG for Tor sharing.
p.s also, very, very nice, to create clearly option to select sorce for internet and target for Tor.

radio24, 2021-06-01:
I don't have an OTG device, and I didn't use one so far.
Could you please elaborate a little more in detail and with an example, which hardware (manufacturer / brand) you want to use in which way?

connected201, 2021-06-01:
also, only RPI 4 and Raspberry Pi Zero W have USB OTG!

radio24, 2021-06-02:
According to the linked article, the device name is **_usb0_**.
TorBox support **_usb0_** - probably that would work out of the box with [using entry 8 in the main menu](https://www.torbox.ch/?page_id=775).
Did you already try to install TorBox on a [Raspberry Pi Zero W](https://www.raspberrypi.org/products/raspberry-pi-zero-w/) with a [Zero Stem](https://zerostem.io/)?

radio24, 2021-06-02:
I just realised that it is probably not as easy as I thought because [entry 8 in the main menu](https://www.torbox.ch/?page_id=775) is only for the situation when the **Internet connection is on _usb0_**. However, in the example using a Raspberry Pi Zero W with a Zero Stem, the Internet is coming most probably from **_wlan0_**. The connection type in this situation would be **_wlan0_<--->_usb0_**, which is not supported (yet) by TorBox. 

I ordered a Raspberry Pi Zero W with a Zero Stem and will follow up your idea on a "TorBox on a stick". Don't expect any quick result; I need to have enough time to look into that.

connected201, 2021-06-02:
Hi, i don't have raspberry pi zero w, only raspberry 4 , but if you realize, it will be great to have possibility to chose source eth0 and share via usb0.

closed 2021-08-05 by connected201