zotil/TorBox #44: Feature request- Add information in maine menu about my external / internal ip

connected201, 2021-06-02:
Hi, it will be nice if in main menu i can see which internal network settings, and external ip, country, city i have right now, like parse from [myip](https://api.myip.com) or another service. Best regards

radio24, 2021-06-02:
We will evaluate if an information page with essential data makes any sense.

On principle both important internal IP addresses 192.168.42.1 and 192.168.43.1 are clearly listed [on our website](https://www.torbox.ch/) and [on the installation flyer](https://www.torbox.ch/wp-content/uploads/2021/04/TorBox-A5-RPI4-040.png). I don't think any other internal network settings are of interest.

If you set up a bridge relay, you will find all necessary information using entry 5 in the ["Defend the open Internet" sub-menu](https://www.torbox.ch/?page_id=913).

External IP addresses have to be looked up from the clients because the traffic from the terminal (for example, for maintenance and update) is not routed through tor, and different clients have different IP addresses, which changes all 10 minutes. Alternatively, you can see all available circuits with main menu entry 1 and pressing the right arrow key once. If you use macOS, then "[IP in menubar](https://www.monkeybreadsoftware.de/Software/IPinmenubar.shtml)" would probably be worth a look.

connected201, 2021-06-02:
also question about  192.168.42.1 and 192.168.43.1, i have case when i want to connect over wifi to have access on device that is connected via eth, so im connecting via wifi and i have ip like 192.168.42.x but my eth device have 192.168.43.x and i cannot connect to 192.168.43.x because im in another subnet, it's possible to make 1 local network?

radio24, 2021-06-02:
I cannot follow. Maybe a sketch would help.

closed 2021-08-05 by connected201