zotil/TorBox #45: How to change DNS server in torbox?

connected201, 2021-06-02:
Hi, Question is, how to safety change DNS server with [Adguard home](https://github.com/AdguardTeam/AdGuardHome) installed locally in torbox?
i want to see and block some DNS queries with Adguard home.
if i correctly understand i need to modify torrc file only?
best regards

radio24, 2021-06-02:
To integrate [Adguard Home](https://github.com/AdguardTeam/AdGuardHome), [Pi-hole](https://pi-hole.net/) or any other DNS filter, you cannot directly redirect DNS requests over tor. Thus you have to change three files: `/etc/iptables.ip4.net`, `set_interfaces` and `set_interfaces_2`:
Search for `udp --dport 53 -j REDIRECT --to-ports 9053` and change it to `udp --dport 53 -j REDIRECT --to-ports 53`. You may configure the DNS filter using port 9053 for its outgoing DNS request to ensure that DNS requests are made through tor again. For that reason, you should add DNSPort 9053 above the DNSPort 192.168.42.1:9053 row in torrc.

To be clear, I didn't test it out - it may work, it may fail. Adguard Home is most likely unscrewing on network tools and configurations, which are specifically configured for TorBox. Furthermore, by adding a DNS filter, you risk degradation of security and anonymity. 

For additional information, see in the FAQ: "[Could you incorporate pi-hole to remove ads?](https://www.torbox.ch/?page_id=112#could-you-incorporate-pi-hole-to-remove-ads)"

connected201, 2021-06-02:
thanks for detailed answer!

closed 2021-06-02 by connected201