zotil/TorBox #50: QUESTION - Why the chosen site to ping is google.com?

nyxnor, 2021-06-06:
Uptime? Availability across the word? Common?
Any problems if changing to debian.org, gitlab.com? or google just blend in to the crowd if using on public space?

radio24, 2021-06-06:
We chose google.com because it is used all over the world, even in authoritarian countries. As you correctly stated, other requirements are uptime and publicity (as more traffic as better).

Based on discussions of a few users, who don't want to use google.com at all, in every file, which checks the online status with a ping or which resolves DNS requests directly through the internet, a variable with the used URL on the top of the script can easily be changed.

We may add an entry in the configuration sub-menu, where a user could change these entries, but we didn't decide on that yet.

nyxnor, 2021-06-06:
Thank you guys for giving always elaborated answers, maybe including `ping_server` in `torbox.lib` to just need to change one file.

radio24, 2021-07-15:
The long-term goal is to torify the local traffic of TorBox as much as possible, but without breaking the captive portal functionality and the possibility to fix a broken tor installation.

Nevertheless, currently, we use open DNS servers whenever torproject.org is used. This is not a silver bullet against blocking mechanisms - we are aware of it - but it overcomes cheap censorship mechanisms (we are testing it under real-world situations). Starting with TorBox v.0.4.2, we will use these open DNS servers more consequent, but at the same time give the user an easy possibility to change the DNS servers. For that reason, we implemented the configuration of the DNS servers into run/torbox.run.

nyxnor, 2021-07-15:
this is the best router ever, thank you.

closed 2021-07-17 by radio24