zotil/TorBox #55: bug removing selected ipv6 bridge

nyxnor, 2021-06-09:
I don't know why this is happening, couldn't figure out the error.
I can only test this for the selected bridge, but `remove if not existent` should be buggy too.
Only happens for the ipv6, commented or uncommented, it is not removed.

nyxnor, 2021-06-09:
Ok, found the bug, but not the solution yet.
Removing `[` and `]` from the ip, let the bridge be removed....

radio24, 2021-06-09:
Probably, in a first step, `[` and `]` have to be identified and then to be changed to` \[ `and `\]` in the variable so that they are identified as a character and not as a delimiter.

nyxnor, 2021-06-09:
Bug is on `bridges_remove_old`, sorry for not metioning, will see what I can use from the above.

radio24, 2021-06-09:
There are three part in that script:
- Remove ALL configured bridges
- Remove only bridges, which do not exist anymore
- Remove only selected bridges"

Which one doesn't remove the ipv6 bridge?

nyxnor, 2021-06-09:
2,3
>I can only test this for the selected bridge, but remove if not existent should be buggy too.

radio24, 2021-06-09:
Probably the problem lies in these two rows:
`sudo grep -v "^${configured_bridges_deactivated[$i]}" ${TORRC} > ${TMP}; sudo mv ${TMP} ${TORRC}`
`sudo grep -v "^${configured_bridges_activated[$i]}" ${TORRC} > ${TMP}; sudo mv ${TMP} ${TORRC}`

I guess grep cannot find and exclude the ipv6 addresses because grep interprets the brackets as a delimiter instead of a character. Therefore, I would try to add the following commands before these two rows:
```
bridge_to_delete="$(<<< "${configured_bridges_deactivated[$i]}" sed -e 's`[][\\/.*^$]`\\&`g')"
sudo grep -v "^$bridge_to_delete" ${TORRC} > ${TMP}; sudo mv ${TMP} ${TORRC}
```

```
bridge_to_delete="$(<<< "${configured_bridges_activated[$i]}" sed -e 's`[][\\/.*^$]`\\&`g')"
sudo grep -v "^$bridge_to_delete > ${TMP}; sudo mv ${TMP} ${TORRC}
```

nyxnor, 2021-06-09:
Ok, try this. Test environment is just the ipv6 bridge, that is why there is no loop.

```
#!/bin/bash

.  lib/torbox.lib
number_of_bridges

i=0
bridge_address=$(cut -d ' ' -f2- <<< ${configured_bridges_deactivated[$i]})
#This is necessary to work with special characters in sed
ORIGINAL_STR="#Bridge ${bridge_address}"
ORIGINAL_STR="$(<<< "${ORIGINAL_STR}" sed -e 's`[][\\/.*^$]`\\&`g')"
ORIGINAL_STR="${ORIGINAL_STR}"
REPLACEMENT_STR="#Bridge $bridge_address"
REPLACEMENT_STR="$(<<< "${REPLACEMENT_STR}" sed -e 's`[][\\/.*^$]`\\&`g')"

sudo sed -i "s/${ORIGINAL_STR}/${REPLACEMENT_STR}/g" ${TORRC}
sudo sed -i "/${REPLACEMENT_STR}/d" ${TORRC}
```

nyxnor, 2021-06-09:
Yours works on a separate script by itself, but not on loop, dont know why, and is much cleaner than mine.

nyxnor, 2021-06-09:
Did not include this ipv6 solution on my PR cause I was unsure, so letting this to you.

radio24, 2021-06-09:
I will finish some work and then go through your PR. This will take some time.
Thanks for your work and the invested time.

radio24, 2021-06-13:
This issue will be fixed with TorBox v.0.4.1, which should be released during the day.

closed 2021-06-13 by radio24