zotil/TorBox #57: security policy - project PGP

nyxnor, 2021-06-10:
There is this solely https://github.com/radio24/TorBox#contact

* Does  the project have PGP?

* If yes, can the releases be signed? 

* Could you write a small security policy? This project is gaining more attention and should the clear how to report security vulnerabilities https://github.com/radio24/TorBox/new/master

Are you radio24 on Tor irc channel? If yes, I would need means to verify that.

leonarzoh, 2021-07-02:
hey @radio24 I send you an email a few minutes ago with a new logo proposal, please have a look when you get a chance. I also would like to request a PGP public key if that's okay with you.

radio24, 2021-07-03:
So far, the project doesn't have a PGP public key; I don't have one, too. 

As far as PGP is concerned, I am ambivalent. Of course, I see the importance, but honestly, PGP is "a pain in the ass". I had offered PGP for emails in another project, but only very few people used this possibility, and then there were problems from time to time, that suddenly emails could not be read anymore and so far. Of course, this ambivalence is perhaps rooted in the fact that my counterparts and I are not working in IT and that the willingness "to play around" with PGP is pretty limited.

I have therefore decided that the mail to the project runs on Protionmail. If someone wants to communicate securely with me, he should also use Protonmail (they offer free accounts). I agree that this could be explicitly noted in the contact details.

radio24, 2021-07-03:
> hey @radio24 I send you an email a few minutes ago with a new logo proposal, please have a look when you get a chance. I also would like to request a PGP public key if that's okay with you.

Hello leonarzoh
Good to hear that you think TorBox is useful. Thank you for the logo proposal - very nice. However, we will stick for the time being with the current logo. Nevertheless, it would be nice if you would share the proposed logo here with other interested people.

leonarzoh, 2021-07-03:
Unfortunately, [PGP user experience](https://latacora.micro.blog/2019/07/16/the-pgp-problem.html) a _pain in the ass_. I agree :) 

Perhaps the best approach for a security policy could be a Warrant Canary on the webpage footer. [EFF coalition drop off the project](https://www.eff.org/deeplinks/2016/05/canary-watch-one-year-later) a few years ago, but a Canary would be great to check if TorBox receives a request for information of any kind by any third parties. [Here](https://ronindojo.io/canary/) an example made by [RoninDojo](https://ronindojo.io/). Anyway, just a suggestion.

nyxnor, 2021-07-03:
@radio24 
> As far as PGP is concerned, I am ambivalent. Of course, I see the importance, but honestly, PGP is "a pain in the ass". I had offered PGP for emails in another project, but only very few people used [...]

The users of this project are probably more familiar with PGP due to using this router. Not everyone knows of course, but this can be taught. The PGP would be more related to verifying the authenticity of the software than to sending e-mails, as I don't see a vulnerability here for now. Also, needs just 2 lines in the readme, 1 to import the sig and 1 to verify the software.

@leonarzoh Canary does not fit this type of project in my view.
* It does not communicate to a server
* It does not have telemetry
* It has few lines of code
* It has only 1 repository to be easily verified
* It has minimal dependencies, just enough to be a router
* Keep reading the commits, always compare the latest version with the one you were currently using to be easier to check the changes https://github.com/radio24/TorBox/compare/v.0.4.0-10.04.2021...v.0.4.1-13.06.2021

radio24, 2021-07-03:
For sure, I will look into it (and also regarding the suggested security policy), when I'm back from my travels (around October/November).

radio24, 2021-07-11:
I added the [public PGP key](https://raw.githubusercontent.com/radio24/TorBox/master/PUBLICKEY.asc) to the TorBox repository. From now on, the releases on GitHub should be signed. I would also add that even before, the image and configuration files linked on the TorBox website can be verified, [using SHA-256](https://www.torbox.ch/?page_id=1128).

@nyxnor Do you have a good example for a security policy?

radio24, 2021-07-14:
Done!

closed 2021-07-14 by radio24