zotil/TorBox #62: signaling newnym

nyxnor, 2021-06-15:
There is many ways to signal newnym
new_ident.sh requires password, this can be cut off

```
sudo -u debian-tor tor-prompt --run 'SIGNAL NEWNYM'
```
`-i 9051` does not need to be mentioned if using the default port.

Source: https://stem.torproject.org/tutorials/down_the_rabbit_hole.html

nyxnor, 2021-07-07:
Restart should still be a menu entry, but give option to just HUP helps too.
Using restarting_tor and including force option to make sure to restart when required

```bash
# restarting_tor(<source script> <force>)
# Used predefined variables: RED, NOCOLOR
# This function restarts Tor if required with 'force', else it will just sighup.
restarting_tor()
{
  SOURCE_SCRIPT=$1
  ACTION=$2
  if [ "${ACTION}" == "force" ]; then
    echo -e "${RED}[+] Restarting Tor!${NOCOLOR}"
    sudo systemctl restart tor &
    echo -e "${RED}[+] DONE! Checking progress - please be patient!${NOCOLOR}"
    echo -e "    Ignore any error messages, just wait..."
    echo -e "    At the end, you should see \"Bootstrapped 100%: Done\"."
    echo -e "    You can leave the progress report with CTRL-C."
  else
    echo -e "${RED}[+] Sending HUP signal to Tor!${NOCOLOR}"
    sudo python3 stem.sighup.py &
    echo -e "${RED}[+] DONE! Checking progress - please be patient!${NOCOLOR}"
    echo -e "    Ignore any error messages, just wait..."
    echo -e "    You can leave the progress report with CTRL-C."
  fi
  echo ""
  sleep 2
  trap "bash $SOURCE_SCRIPT; exit 0" SIGINT
  sudo tail -f -n 0 /var/log/tor/notices.log
}
```

```python3
from stem import Signal
from stem.control import Controller

with Controller.from_port(port = 9051) as controller:
  controller.authenticate(password=password)
  controller.signal(Signal.HUP)
```

radio24, 2021-07-14:
With [commit dbd551c](https://github.com/radio24/TorBox/commit/dbd551cae1304b7b8c4f8a34e6ed3dc360de403b), the new_ident script is replaced by sudo -u debian-tor tor-prompt --run 'SIGNAL NEWNYM'

radio24, 2021-07-15:
Regarding the proposal to implement to HUP tor instead of restarting it, I did a few tests during the last few days. So far, I don't see a significant advantage to distingue between a "real" restart and a "reload" of the configuration. On the contrary, the HUP variant gives less information about reloading. Also, I think that with restarting tor, I'm on the safer side, even if it needs more time on some occasions.

nyxnor, 2021-07-15:
> I don't see a significant advantage to distingue between a "real" restart and a "reload" of the configuration.

Did you test with bridges? With bridges it has significant difference cause they take more time to bootstrap.

> On the contrary, the HUP variant gives less information about reloading.

It gives then necesssaary info of what it does, reading the config files and data dir

>I'm on the safer side, 

Also, to be on the safer side there is
`sudo systemctl reload-or-restart`, it will reload the service if it is active, it it is failed or stopped it will restart it.

* Plus I do recommend leaving both options. Restart and Reload, but only restart when choosing that option on the menu, else when editing the torrc, just reload.

radio24, 2021-07-15:
> > I don't see a significant advantage to distingue between a "real" restart and a "reload" of the configuration.
> 
> Did you test with bridges? With bridges it has significant difference cause they take more time to bootstrap when restarting and no need to do so when reloading.

Yes, I tested it with OBFS4 bridges but ran into problems. It didn't seem to work, but I couldn't follow the log because of the reduced log entries. When I restarted tor with the same configuration, I had no problem, and I could follow the log entries if the bridge is used correctly. So, I thought that maybe HUP will not work with OBFS4 bridges.

closed 2021-07-15 by radio24