zotil/TorBox #64: BSD platform

nyxnor, 2021-06-29:
Yes, I've read that support to other platforms are experimental, but they are still linux, this is BSD and can contribute.

**Why?** 
[Linux dominates](https://metrics.torproject.org/platforms.html), BSD is at the bottom, need some risk minimization if there is some exploit on the kernel

**Image where?**
[FreeBSD](https://wiki.freebsd.org/arm/Raspberry%20Pi) and [OpenBSD](https://www.openbsd.org/arm64.html)

**Ok, but how?** 

* FreeBSD
  * [Bridge](https://community.torproject.org/relay/setup/bridge/freebsd/)
  * [normal relay](https://community.torproject.org/relay/setup/guard/freebsd/)
  * [relay guide](https://github.com/seth586/guides/blob/master/FreeNAS/tor_relay/README.md)

* OpenBSD
  * [Bridge](https://community.torproject.org/relay/setup/bridge/openbsd/)
  * [Guard](https://community.torproject.org/relay/setup/guard/openbsd/)

**Can I help?**
Never tried BSD, but there is always time for learning.

**Question**
Viable? Any plans for that?

radio24, 2021-06-29:
This is a nice idea. However, because a BSD port is more complicated than supporting other Linux distributions, currently, I cannot care about a BSD port. 

I think it would be the easiest way for you to fork TorBox into a new repository, for example, TorBox-BSD.

nyxnor, 2021-07-16:
Ok, I like this idea too much to close it, but for I do not have time at this moment, so will close this for now.

radio24, 2021-07-16:
I put it on the long-term to-do list.

nyxnor, 2021-12-08:
I have been running OpenBSD for a week now. It is great, default shell is ksh, not bash but it can run bash, it is just not installed by default. The bad thing portability, would have to prepend every call script with `bash` to specify the shell or call the script with `./script.sh` to follow the shebang.

But besides being using bash or not, it does not work with the same tools of `sed`, `awk`, `grep` if they are not POSIX. This is not "hard" to do but takes time to learn, read the manuals. OnionJuggler is POSIX and running on OpenBSD (minus authentication cause it needs a missing package and vanguards because it is running on systemd not openrc).

So it would be a matter of taking some time to port script by script, one at a time.
Also there could be missing packages, debian is full of them.

The good thing of doing this is that you are not tied to bash and debian derived systems, you can run other kernels, other operating systems, so it is a great thing.

> I think it would be the easiest way for you to fork TorBox into a new repository, for example, TorBox-BSD.

It would be unmaintainable because it is not gonna be little patches or simple fixes, it you be on every script.

Anyway, I got this idea because it makes torbox portable on the software level, but I cannot do it alone, only if some guy called @radio24 would be interested in that. Aiming for december next year to finish.

radio24, 2021-12-08:
Even my online servers are running on FreeBSD, and the portation of TorBox to BSD would be admirable, it is not a priority of mine. TorBox started as a Raspberry Pi OS only project, with all the portability problems. Nevertheless, it also works on Debian and Ubuntu. My next intermediate goal would be to port it to Armbian (and/or maybe DietPi) because of the wide supports of different SBCs.

As you said, the scripts have to be ported one after another for BSD, which is time-consuming. These "new" scripts not only have to be changed correctly, but the system also has to be tested. Because of my everyday use of TorBox, the current implementation on Raspberry Pi OS is very detailed tested - this has to be redone for a BSD version. Unfortunately, feedback from the community is a little bit rare.

Regarding my priorities, I don't think that I will have much time during 2023, so I will concentrate on the essential points, which are the integration of TFS and TCS as well as SSH / SFTP / XMP over Onion domains, the implementation of the creation of Onion v3 vanity addresses, the possibility to optionally install WordPress and PHP, the replacement of iptables by nftables, the possible replacement of hostapd, IPv6 support, replacement of Shellinabox, detailed testing of the side effects using randomize MAC addresses, the implementation of more countermeasures (deauthentication ...) and more connection possibilities.

I agree with you that a BSD version of TorBox would be nice, but from the time I have available and my priority list, a BSD portation has to wait.

nyxnor, 2021-12-09:
No problem, see issue again on 2023.

Just not that porting to BSD is porting to a Unix system, meaning it would also work on other Linux distros.

About the goals for 2022, please open an I issue so I can provide information.

closed 2021-12-09 by nyxnor