zotil/TorBox #68: TorBox Tails

247Hackman, 2021-07-02:
Is there a way I can configure torbox with tails

nyxnor, 2021-07-02:
Never tried pointing Tails to an external port but:

radio24, 2021-07-03:
Why should someone use TorBox and Tails together? If you use Tails, all data are already going through Tor. To prevent mistakes, Tails automatically blocks applications if they try to connect to the Internet without Tor (see [here](https://tails.boum.org/about/index.en.html)). 

Combining Tails with TorBox would mean that your data would go twice through Tor; meaning the data stream goes through two circuits, which means that you get six hops instead of three (for more details, see [here](https://www.torbox.ch/?page_id=112#when-im-connected-with-torbox-and-use-the-tor-browser-on-one-of-the-clients-isnt-that-a-risk-for-my-securityanonymity-tor-over-tor-scenario)). Even we don't think this is a security risk, it doesn't have any advantages, in contrary your overall network speed will be most likely worse than usual.

If you really want to use TorBox and Tails together, then no particular configuration should be necessary. For Tails, TorBox is nothing else than a wifi- or cable-router.

closed 2021-07-15 by radio24