zotil/TorBox #69: New logo proposal

leonarzoh, 2021-07-03:
Hi all. Just would like to share a new logo proposal with everyone, as suggested by @radio24.
Basically, it's a purple box using the onion icon. I can create alternative versions, perhaps using green or something like that.  I have the vector version (svg, ai etc) on Figma, so I can share the file if someone would like to fork it. 

[image]

radio24, 2021-11-19:
I reopen this contribution from [leonarzoh](https://github.com/leonarzoh). 
Probably, I will use it for the splash screen shown when a folder is shared through Onion Services, integrated with the next version of TorBox.

radio24, 2021-11-19:
Coming soon ....
[image]

nyxnor, 2021-11-19:
wowwww

closed 2021-11-19 by radio24