zotil/TorBox #70: bridges scripts cleanup and sourcing from torbox.lib

nyxnor, 2021-07-07:
URL, ,enu parameters (height, width), terminal colors, $TORRC, activation, deactivation, removal of bridges still remain inside `bridges_*` , `meek-azure`, `snowflake` scripts, although most of the functions are already included in `torbox.lib`.
Can I remove this from inside scripts and move all of them to the lib?
This avoids changing multiple scripts when modification is needed, lib will be canonical for all scripts, as it should be being a library. I saw you already started doing this kinda of changes recently, so it is in the right track.

Approval to patch and PR this?

radio24, 2021-07-07:
Of course, patch it and make a PR. Of course, I will test it a little bit and probably after the 23rd of July, when I‘m back home for 10 days, I integrate it to the master or to another branche.

Your contribution is always highly appreciated.

nyxnor, 2021-07-07:
Is this correct? 
https://github.com/radio24/TorBox/blob/2ce64182537adf615d1bb8026d058ee9af88bc74/lib/torbox.lib#L369

It deactivates obfs4 bridges if user answers no to restart at that moment, why?

radio24, 2021-07-07:
You are absolutely right.
Because that is a serious error, let me fix it right now in the master branch.

nyxnor, 2021-07-07:
>The reason is that by removing the two steps to activate the OBFS4 bridges in TorBox, I will avoid the situation that users configure the OBFS4 bridge functionality but don't use them.

I don't see the problem with that. If user still needs Tor at that moment and has already chosen the bridges, he can just restart tor later to activate it. It is even better this way than clicking "no" to restart and deactivate bridges, as this take some time.

>An alternative option could be to restart tor without asking first. On the other hand, especially for inexperienced users, it is good to be informed about the next step.

Will loose connection and maybe it is needed at that moment, giving an option is better. It also helps a lot testing to not loose time waiting for tor to restart.

nyxnor, 2021-07-07:
I was working with this today for `torbox.lib`:

```bash
activate_bridges_default_lines(){
  sudo sed -i "s/^#UseBridges/UseBridges/g" ${TORRC}
  sudo sed -i "s/^#UpdateBridgesFromAuthority/Update  BridgesFromAuthority/g" ${TORRC}
}

deactivate_bridges_default_lines(){
  sudo sed -i "s/^UseBridges/#UseBridges/g" ${TORRC}
  sudo sed -i "s/^UpdateBridgesFromAuthority/#Update  BridgesFromAuthority/g" ${TORRC}
}

# bridges_function <source script> <activate|deactivate> <obfs4|meek_lite|snowflake|all>
bridges_function(){

  SOURCE_SCRIPT=${1}
  ACTION="${2}"
  TRANSPORT="${3}"

  if [ "${TRANSPORT}" == "meek_lite" ]||[ "${TRANSPORT}" == "obfs4" ]; then
    TRANSPORTSTRINGa="meek_lite,obfs4"
    if [ "${TRANSPORT}" == "meek_lite" ]; then
      TRANSPORTSTRINGb="meek_lite"
    else
      TRANSPORTSTRINGb="obfs4"
    fi
  elif [ "${TRANSPORT}" == "snowflake" ]; then
    TRANSPORTSTRINGa="snowflake"
    TRANSPORTSTRINGb="snowflake"
  fi

  if [ "${ACTION}" == "activate" ]; then
    if [ "${TRANSPORT}" == "obfs4" ]; then
      number_of_bridges
      if [ $number_configured_bridges_activated -gt 0 ]; then
        deactivating_bridge_relay
        activate_bridges_default_lines
        sudo sed -i "s/^#ClientTransportPlugin ${TRANSPORTSTRINGa}/ClientTransportPlugin ${TRANSPORTSTRINGa}/g" ${TORRC}
        sudo sed -i "s/^ClientTransportPlugin snowflake /#ClientTransportPlugin snowflake /g" ${TORRC}
        sudo sed -i "s/^Bridge meek_lite/#Bridge meek_lite/g" ${TORRC}
        sudo sed -i "s/^Bridge snowflake/#Bridge snowflake/g" ${TORRC}
        needRestart=1
      else
        deactivate_bridges_default_lines
        sudo sed -i "s/^ClientTransportPlugin ${TRANSPORTSTRINGa}/#ClientTransportPlugin ${TRANSPORTSTRINGa}/g" ${TORRC}
        sudo sed -i "s/^Bridge ${TRANSPORTSTRINGb} /#Bridge ${TRANSPORTSTRINGb} /g" ${TORRC}
        sudo sed -i "s/^Bridge meek_lite/#Bridge meek_lite/g" ${TORRC}
        sudo sed -i "s/^Bridge snowflake/#Bridge snowflake/g" ${TORRC}
        needRestart=1
      fi
    elif [ "${TRANSPORT}" == "meek_lite" ]||[ "${TRANSPORT}" == "snowflake" ]; then
      deactivating_bridge_relay
      activate_bridges_default_lines
      sudo sed -i "s/^#ClientTransportPlugin ${TRANSPORTSTRINGa}/ClientTransportPlugin ${TRANSPORTSTRINGa}/g" ${TORRC}
      sudo sed -i "s/^#Bridge ${TRANSPORTSTRINGb}/Bridge ${TRANSPORTSTRINGb}/g" ${TORRC}
      sudo sed -i "s/^Bridge obfs4/#Bridge obfs4/g" ${TORRC}
      if [ "${TRANSPORT}" == "meek_lite" ]; then
        sudo sed -i "s/^Bridge snowflake/#Bridge snowflake/g" ${TORRC}
      elif [ "${TRANSPORT}" == "snowflake" ]; then
        sudo sed -i "s/^Bridge meek_lite/#Bridge meek_lite/g" ${TORRC}
      fi
      needRestart=1
    fi
  elif [ "${ACTION}" == "deactivate" ]; then
    if [ "${TRANSPORT}" == "all" ]; then
      deactivate_bridges_default_lines
      sudo sed -i "s/^ClientTransportPlugin meek_lite,obfs4/#ClientTransportPlugin meek_lite,obfs4/g" ${TORRC}
      sudo sed -i "s/^ClientTransportPlugin snowflake/#ClientTransportPlugin snowflake/g" ${TORRC}
      sudo sed -i "s/^Bridge obfs4/#Bridge obfs4/g" ${TORRC}
      sudo sed -i "s/^Bridge meek_lite/#Bridge meek_lite/g" ${TORRC}
      sudo sed -i "s/^Bridge snowflake/#Bridge snowflake/g" ${TORRC}
      needRestart=1
    else
      deactivate_bridges_default_lines
      sudo sed -i "s/^ClientTransportPlugin ${TRANSPORTSTRINGa}/#ClientTransportPlugin ${TRANSPORTSTRINGa}/g" ${TORRC}
      sudo sed -i "s/^Bridge ${TRANSPORTSTRINGb}/#Bridge ${TRANSPORTSTRINGb}/g" ${TORRC}
      needRestart=1
    fi
  fi

  if [ ${needRestart} -gt 0 ]; then
    INPUT=$(cat text/restart-tor-bridges-text)
    if (whiptail --title "TorBox - INFO" --defaultno --no-button "NO - DON'T (RE)START" --yes-button "YES - (RE)START" --yesno "$INPUT" $MENU_HEIGHT_25 $MENU_WIDTH); then
      restarting_tor $SOURCE_SCRIPT
    fi
  fi

}
```

radio24, 2021-07-07:
Hum, at first sight, it seems to be more complicated than before. For me, the scripts and functions must be simple and, therefore, easy to maintain. I am not too fond of scripts and functions which does all certain things, and whenever someone changes something, he will break something else. KISS: keep it simple and stupid.

nyxnor, 2021-07-07:
KISS won this time, but I'LL BE BACK.
Could you respond to this? https://github.com/radio24/TorBox/issues/70#issuecomment-875485730 . Or your point remains that bridges should be commented if not restarting tor when prompted. I still don't think it should be that way.
```bash
activate_default_lines_bridges(){
  sudo sed -i "s/^#UseBridges/UseBridges/g" ${TORRC}
  sudo sed -i "s/^#UpdateBridgesFromAuthority/UpdateBridgesFromAuthority/g" ${TORRC}
}

deactivate_default_lines_bridges(){
  sudo sed -i "s/^UseBridges/#UseBridges/g" ${TORRC}
  sudo sed -i "s/^UpdateBridgesFromAuthority/#UpdateBridgesFromAuthority/g" ${TORRC}
}

deactivate_meek_bridges(){
  sudo sed -i "s/^ClientTransportPlugin meek_lite,obfs4/#ClientTransportPlugin meek_lite,obfs4/g" ${TORRC}
  sudo sed -i "s/^Bridge meek_lite /#Bridge meek_lite /g" ${TORRC}
}

deactivate_snowflake_bridges(){
  sudo sed -i "s/^ClientTransportPlugin snowflake/#ClientTransportPlugin snowflake/g" ${TORRC}
  sudo sed -i "s/^Bridge snowflake /#Bridge snowflake /g" ${TORRC}
}

# deactivate_obfs4_bridges()
# Dectivates OBFS4 if the number of activated obfs4 bridge entries in torrc is 0
deactivate_obfs4_bridges(){
  sudo sed -i "s/^ClientTransportPlugin meek_lite,obfs4/#ClientTransportPlugin meek_lite,obfs4/g" ${TORRC}
  sudo sed -i "s/^Bridge obfs4 /#Bridge obfs4 /g" ${TORRC}
}

# activate_obfs4_bridges(<source script>)
# Activates OBFS4 if the number of activated obfs4 bridge entries in torrc is >0
activate_obfs4_bridges()
{
  SOURCE_SCRIPT=$1
  number_of_bridges
  if [ $number_configured_bridges_activated -gt 0 ]; then
    deactivating_bridge_relay
    deactivate_meek_bridges
    deactivate_snowflake_bridges
    activate_default_lines_bridges
    sudo sed -i "s/^#ClientTransportPlugin meek_lite,obfs4/ClientTransportPlugin meek_lite,obfs4/g" ${TORRC}
  else
    deactivate_default_lines_bridges
    deactivate_obfs4_bridges
  fi

  INPUT=$(cat text/restart-tor-bridges-text)
  if (whiptail --title "TorBox - INFO" --defaultno --no-button "NO - DON'T (RE)START" --yes-button "YES - (RE)START" --yesno "$INPUT" $MENU_HEIGHT_25 $MENU_WIDTH); then
    restarting_tor $SOURCE_SCRIPT
  fi
}

# activate_meek_bridges(<source script>)
activate_meek_bridges()
{
  SOURCE_SCRIPT=$1
  deactivating_bridge_relay
  activate_default_lines_bridges
  deactivate_obfs4_bridges
  deactivate_snowflake_bridges
  sudo sed -i "s/^#ClientTransportPlugin meek_lite,obfs4/ClientTransportPlugin meek_lite,obfs4/g" ${TORRC}
  sudo sed -i "s/^#Bridge meek_lite /Bridge meek_lite /g" ${TORRC}

  INPUT=$(cat text/restart-tor-bridges-text)
  if (whiptail --title "TorBox - INFO" --defaultno --no-button "NO - DON'T (RE)START" --yes-button "YES - (RE)START" --yesno "$INPUT" $MENU_HEIGHT_25 $MENU_WIDTH); then
    restarting_tor $SOURCE_SCRIPT
  fi
}

activate_snowflake_bridges(){
  SOURCE_SCRIPT=$1
  deactivating_bridge_relay
  activate_default_lines_bridges
  deactivate_obfs4_bridges
  deactivate_meek_bridges
  sudo sed -i "s/^#ClientTransportPlugin snowflake/ClientTransportPlugin snowflake/g" ${TORRC}
  sudo sed -i "s/^#Bridge snowflake /Bridge snowflake /g" ${TORRC}

  INPUT=$(cat text/restart-tor-bridges-text)
  if (whiptail --title "TorBox - INFO" --defaultno --no-button "NO - DON'T (RE)START" --yes-button "YES - (RE)START" --yesno "$INPUT" $MENU_HEIGHT_25 $MENU_WIDTH); then
    restarting_tor $SOURCE_SCRIPT
  fi
}
```

radio24, 2021-07-07:
> I don't see the problem with that. If user still needs Tor at that moment and has already chosen the bridges, he can just restart tor later to activate it. It is even better this way than clicking [...]

My concern is that if someone is not restarting tor and forget about it, that this can have some side effects with other functions. The main question is, why should someone be interested in activating bridges and not restarting tor? The most likely answer l have is that he didn't want to activate OBFS4 bridges in the first place, or he doesn't know what he is doing. If we stick to the save side, it shouldn't be possible to have activated bridge rows in torte, which are not used. Further, I don't see any benefit to allowing that.

However, agree with you that restarting tor without asking wouldn't be a good solution.

nyxnor, 2021-07-22:
just to provide some order and expectation:
* finish Vanguards
* Installation script on a censored network
* move scripts to torbox.lib

radio24, 2021-07-22:
- Vanguard: tell me when you do your latest changes then I will start to integrate the pull request.
- Installation script: I will wait until Sunday, 25 JUL 21, and then step into it. I need these scripts to build/test the image files, which I have to start next week. My goal for the v.0.4.2 release is the end of July.
- Move scripts to torbox.lib: will not be integrated into v.0.4.2 because of lack of time.

closed 2021-08-02 by radio24