zotil/TorBox #72: add Vanguards

nyxnor, 2021-07-09:
_Even after deployment of the new v3 onion service protocol, the attacks facing onion services are wide-ranging, and still require more extensive modifications to fix in Tor-core itself._

_Because of this, we have decided to rapid-prototype these defenses in a controller addon in order to make them available ahead of their official Tor-core release, for onion services that require high security as soon as possible._

This is for the controller, so I think it is good to be added.
I have a script that runw from git, creates a systemd service for vanguards to execute the python script. I still want to enhance it to be a menu option to show logs, install, remove, start, stop, but that is the last part after all things are ready.
https://github.com/mikeperry-tor/vanguards#running-this-addon-directly-from-git
Would be good to enhance the[log check](https://github.com/radio24/TorBox/issues/71#issue-938913693) with [vanguards logs](https://github.com/mikeperry-tor/vanguards/blob/master/README_SECURITY.md#monitor-your-service)

Important part:
Want this?
Default installation or optional installation?

radio24, 2021-07-10:
This looks interesting. It is very easy to enhance the [log check](https://github.com/radio24/TorBox/issues/71#issue-938913693) with [vanguards logs](https://github.com/mikeperry-tor/vanguards/blob/master/README_SECURITY.md#monitor-your-service) because [log_check.py](https://github.com/radio24/TorBox/blob/automatisation-test/log_check.py) is working with al kind of logs.

**Want this?** -> YES
**Default installation or optional installation?** -> First for testing as an optional installation, later maybe as default.

nyxnor, 2021-07-22:
I will add commit hash to not run directly from the git repo, but to reset til specified commit.
Also will add `sudo -u debian-tor git clone`.

radio24, 2021-07-25:
I integrated the pull request and ran some first tests. I guess the fact that Raspberry Pi OS and Debian is a little bit different complicates some things. Here are some questions and issues with the installation of Vanguards:
- `sudo -u debian-tor reset --hard ${VANGUARDS_COMMIT_HASH}` -> Necessary? Why not just grab the latest version of the master branch?
- `sudo mv vanguards /var/lib/tor/` -> Error! What do you want to copy to `/var/lib/tor/`? The entire cloned vanguards repository? Is that really necessary?
- `sudo cp /var/lib/tor/vanguards/vanguards-example.conf /var/lib/tor/vanguards/vanguards.conf` -> should be in `/etc/tor/`

I think the installation of Vanguards could be moved to the installation script, but the activation shouldn't be by default yet. Nevertheless, this could be a little bit tricky with Ubuntu.

nyxnor, 2021-07-25:
1. VANGUARDS_COMMIT_HASH
* Depends on your security level
* If github user is compromised and malicious commit is merged mainstream, this could be an attack vector. Setting commit hash makes the developer review (or incite to review) the code till that commit for good security measure.
2. mv vanguards /var/lib/tor/
* Yes, the entire cloned repository.
* I don't know which file should not be included in the folder, so I move everything. Also, easier to `git fetch`.
3. vanguards.conf should be in /etc/tor/
* Here you are right, the systemd is searching for that inside /etc/tor. That is the path for the deb package of vanguards, but it could be in the current working directory. From the [README](https://github.com/mikeperry-tor/vanguards#configuration):
_Configuration files can be specified on the command line. The default is to read vanguards.conf from the current working directory. If the environment variable $VANGUARDS_CONFIG is set, the config file will be read from the file specified in that variable._

Thank you for the review, learned a lot and seeing how you implemented the service and menu, will be more familiarized with doing the TorBox design next time.

https://github.com/radio24/TorBox/blob/86de2c6236dc3ef70ff9ce0592e30b3fe25fe940/menu-vanguards#L101
Maybe if `-d $VANGUARD_DIR` exists, then just `git fetch` and restart the service to update?

closed 2021-08-02 by radio24