zotil/TorBox #73: is it possible to change the distribution mechanism of the OBSF4 Bridge on Moat?

DEC-entralized, 2021-07-12:
is it possible to change the distribution mechanism of the OBSF4 Bridge on Moat? I only ever see HTTPS but then my bridge is only distributed via the website and not automatically when clients request.
[image]

radio24, 2021-07-13:
You can try to add „BridgeDistribution moat“ to /etc/tor/torrc. However, the [tor manual](https://2019.www.torproject.org/docs/tor-manual.html.en) states the following:

_If set along with BridgeRelay, Tor will include a new line in its bridge descriptor which indicates to the BridgeDB service how it would like its bridge address to be given out. Set it to "none" if you want BridgeDB to avoid distributing your bridge address, or "any" to let BridgeDB decide. (Default: any) 
Note: as of Oct 2017, the BridgeDB part of this option is not yet implemented. Until BridgeDB is updated to obey this option, your bridge will make this request, but it will not (yet) be obeyed._

[According to a discussion amongst developers on GitLab](https://gitlab.torproject.org/tpo/core/tor/-/issues/40321), it may be that it is again possible to request (or even to enforce) in which "[basket](https://bridges.torproject.org/info?lang=en) "("https", "email" and "moat") your OBFS4 bridge relay should fall. Nevertheless, I thought that with the default ("any "), an OBFS4 bridge relay has to run stable for quite a time before the bridge distribution mechanism changes to "moat". Unfortunately, I couldn't find anything about this aspect anymore.

DEC-entralized, 2021-07-13:
> You can try to add „BridgeDistribution moat“ to /etc/tor/torrc. However, the [tor manual](https://2019.www.torproject.org/docs/tor-manual.html.en) states the following:
>
> _If set along with [...]

Also ich habe bereits mehrere Bridges aufgesetzt mit torbox und dann hat es immer eine weile gedauert bis ein Verteilungsmechanismus angezeit wurde. Entweder wurde HTTPS oder MOAT angezeigt. E-MAIL hatte ich noch nicht.
Ich habe lust zu versuchen einen Menüeintrag zum Defend the open internet Tab zu coden so das man nicht in die config dafür muss bzw man auch daran erinnert wird seinen Verteilungsmechanismus zu wählen. Ich würde es dann per pullrequest hin zu fügen. Das würde mir viel freude bereiten und wäre ein toller ansporn etwas coden zu lernen :) Wenn du das möchtest?

radio24, 2021-07-13:
Feel free to contribute. The new option should be integrated / combined with the other, already existing options ([menu entry 3 in the sub-menu](https://www.torbox.ch/?page_id=913)).

radio24, 2021-07-20:
With commit [#7a15a3417458d11a500390f6a798dd17e174e5bc](https://github.com/radio24/TorBox/commit/7a15a3417458d11a500390f6a798dd17e174e5bc) the distribution method can be configured. It will be helpful if @DEC-entralized could test it out. It will be released with the next TorBox release (v.0.4.2).

DEC-entralized, 2021-07-21:
Hello, I'm sorry that I didn't manage to take care of it in time.
I will test it as soon as I know that my bridge can be reached and used.
I generally can't get an OBSF4 Bridge to work ... I keep getting SOCKS5 errors when I try to connect my TorBox wifi PI to my OBSF4 Bridge PI. I just activate my own separate OBSF4 bridge in the wifi pi and then my wifi pi never manages to build a circuit. Other people have never been on my bridge either. After 3 days I still just uploaded 16 mb and downloaded about 40 mb. I released both 4235 and 443 for the bridge in my newly bought Fritzbox 7590AX. I bought the Fritzbox especially for it, only that I can make more settings. it still doesn't work ... Even after several days it still says: I've made 0 circutes in last 6 hours, etc.
Unfortunately, I was very desperate so I couldn't take care of the MOAT function. I'm sorry.
I will describe my problem later or describe the days in more detail in a new issue. with logs then. I just have no logs since I uninstalled the bridge yesterday because I wanted to set it up again for the x time. It is very difficult to understand what all the messages inside the bridge mean ...

radio24, 2021-07-21:
According to your first post, your bridge can be reached and should work. This doesn't mean that you will immediately see the use of your bridge. It can take several days or weeks until you see an advertised bandwidth and a consistent set of users (see [here](https://www.torbox.ch/?page_id=913)). Don't get discouraged if you don't see user connections right away (see also [here for more details](https://blog.torproject.org/lifecycle-new-relay)).

However, you should be able to use your bridge for your second TorBox and/or also for the TorBrowser. Did you try to use your bridge for TorBrowser on a computer in another network without TorBox in between? When you download and start the TorBrowser 10.5.x, go to the Tor Network Settings, choose "Provide a Bridge", and enter your Bridge information.

DEC-entralized, 2021-07-21:
Hi, I'll try again with a gate browser to connect to my Bridge PI.
At the same time I will test the new 0.4.2 branch.
I'm just about to set it on.

radio24, 2021-07-21:
> to understand. If I don't have a fixed public ip, it doesn't matter or do I have to generate a fixed ip via ddns?

You don't need a public IP, as long as your provider supports the port forwarding. However, the IP distributed with the bridge address has to be fixed. In the example below, the path to 187.199.36.90 has to be clear, the OBFS4 bridge relay and the port 6667 has to be accessible on that IP address:

obfs4 187.199.36.90:6667 03421E8462B2446FAA5D0F632C9B349C6C58433A cert=nnzsKeikJZ2foFyIzu3TixukCUvoyym2p0M2bxjlCL28C7y1ieb1yO0ohmCGO9i07aDBBA iat-mode=0

I don't think DDNS can help you in any way because it has nothing to do with domain name resolution. Also, if the tor log tells you that your ports are reachable, the [TCP reachability test](https://bridges.torproject.org/scan/) is positive, and [Tor Metrics](https://metrics.torproject.org/rs.html) tells you that your relay is online, there is not more to do.

> And does that also work with a dual stack internet connection?

I cannot answer that question because I didn't have the opportunity to test this.

DEC-entralized, 2021-07-21:
So I added the new files to my bridge. But there is no menu entry for the distribution mechanism mode. I have now done it via the advanced config. have to wait and see what relaysearch says my new bridge is not there yet. my ports 443 and 4235 are attainable. now also with ipv6.

DEC-entralized, 2021-07-21:
So my bridge is now also available for my wifi torbox PI. I'm celebrating right now :P  I made the mistake of adding the bridge to give the certificate .... Grrrr xD

radio24, 2021-07-21:
> So I replaced the new files from the 0.4.2 branch with those in my bridge. But there is no menu entry for the distribution mechanism mode. I have now done it via the advanced config. have to wait and see what relaysearch says my new bridge is not there yet. my ports 443 and 4235 are attainable. now also with ipv6.

The selection of the distribution mechanism mode is shown when you use menu entry 3 "Check and/or change the configuration". However, at the moment you have to re-enter all the configuration settings - I guess I can do it better, and will try it tomorrow.

radio24, 2021-07-22:
New on [menu-config](https://github.com/radio24/TorBox/blob/fd26c895adce139dbd53cb093c1a22fa5e8aaba0/menu-config)
- The configurations can also be changed when the OBFS4 bridge relay is running. No deactivation before and activation after the changes are necessary anymore.
- Pressing ENTER in the configuration dialogue is not setting the default values but the latest used ones.

radio24, 2021-08-02:
Implemented with TorBox version 0.4.2!

closed 2021-08-02 by radio24