zotil/TorBox #78: Feature requests

darkza, 2021-07-30:
Add client mode for internet input and
Tor dhcpd output.

Wlan0 raspberry onboard> dhcpcd from ssid1 (normal iNet Input)
Wlan1 usb wlan > clientmode from ssid2 with fixed ipadress and dhcpd server

Why?
I have multiple accespoints in my home and like to use vlan2 and ssid2 for tor only without have to be in range of the raspberry

radio24, 2021-07-30:
If I understand you right, you want all IPs from TorBox (iNet and client) given by one DHCP server (probably your iNet router)?

darkza, 2021-07-30:
No.
I have a standard router that gives the torboxclient "internet" and dhcp.
So torbox gets his wlan0 client.
All fine, pretty normal usecase.

What i can't do with our Script is add wlan1 in clientmode to an other
accespoint. Also torbox wlan1 has to act as dhcp server for this accespoint.

Greetings

radio24, 2021-07-30:
Hmmm
The TorBox is already acting als dhcp server for all connected clients.
Check out the etc/dhcp/dhcpd.conf in the torbox folder:
```
# This configuration file was modified by TorBox

default-lease-time 600;
max-lease-time 7200;
ddns-update-style none;
authoritative;

subnet 192.168.42.0 netmask 255.255.255.0 {
range 192.168.42.10 192.168.42.19;
option broadcast-address 192.168.42.255;
option routers 192.168.42.1;
option domain-name "local";
option domain-name-servers 192.168.42.1;
}

subnet 192.168.43.0 netmask 255.255.255.0 {
range 192.168.43.10 192.168.43.19;
option broadcast-address 192.168.43.255;
option routers 192.168.43.1;
option domain-name "local";
option domain-name-servers 192.168.43.1;
}
```

darkza, 2021-07-30:
Nein versteh ich schon.
Bei der Einrichtung fragt er ja wo kommt das internet her das passt alles
da sag ich ihm wlan0 clientmode dhcpcd.

Wenn ich jetzt aber die wlan1 einrichten will kann ich nur einen "Hotspot"
mit dieser einrichten.

Ich will aber lieber das diese im clientmodus läuft und sich mit dem
fremden accespoints verbindet ganz normal mit einer fixen ip Adresse. Dhcpd
soll diese auch schicken.

Sinn dahinter ist das jetzt alle Smartphones die sich mit dem accespoint
verbinden automatisch und ausschließlich über die torbox ins internet gehen.

Lg

radio24, 2021-07-31:
Ehrlich gesagt, habe ich keine Ahnung, wie ich das konfigurationsmässig integrieren könnte.

closed 2021-08-01 by radio24