zotil/TorBox #80: Manage onion services

nyxnor, 2021-08-03:
https://github.com/nyxnor/raspiblitz/blob/tor-patch/home.admin/config.scripts/tor.onion-service.sh

This can be adapted to read existent folders loop in the dir to find what services exist and place them in the whiptail menu

https://stackoverflow.com/a/2108296
```bash
for dir in /var/lib/tor/services/*/     # list directories in the form "/tmp/dirname/"
do
    dir=${dir%*/}      # remove the trailing "/"
    echo "${dir##*/}"    # print everything after the final "/"
done
```

Services need to be created inside /var/lib/tor/services/ dir for this loop to work, it is better this way than naming them hs_*.

Currently I have not create a menu for this, but I should improve more the services script before designing that.

nyxnor, 2021-08-05:
Let's say I have a set of hidden services I want to add a set of clients,
I can tick in a checklist the services I want to add clients, as well as well as add multiple clients at the same time, separated by a comma like this https://github.com/radio24/TorBox/blob/eb88bba0744b8ee0ec9bab6f232ce1faa4b16420/bridges_activate_old#L163
And if I loose trust a set of clients and want to remove them from the authorized list of multiple services, this should be done too.

Some scripts are required to be run with the `debian-tor` user, as it reads the `/var/lib/tor`to check if service or client exists. I don't think there is a way to avoid that.

connected201, 2021-08-06:
also Please Add option to import and export hidden services and torrc file from home directory, because directly to /var/lib/tor cannot copy files from scp an example, and export too, need special rights., for my is complicated, because, need to edit torrc file.

nyxnor, 2021-08-06:
@connected201 Thank you for this. 
import and export functions will be integrated to preserve onion addresses after flashing the sd card to update major versions of torbox, as well as importing previous services data from another computer.

Process should be to `tar` the folders `/var/lib/tor/onion_auth` and `/var/lib/tor/services`, define remote host (ip, user, path) to scp (maybe store these variables).
A whiptail inputbox to register data and whiptai menu to click to export or import data.

nyxnor, 2021-08-06:
Tasks
- [ ] Error handling, skip if looping, not exit in every situation
- [ ] when renewing addresses, save clients name before deleting all auths and replace them with the new data with the same name
- [ ] mv part of the menu-onion-credentials, menu-onion-auth, menu-onion-renew, menu-onion-create, menu-onion-delete to be inside tor.onion-service.sh. Each menu has for now 100 lines each (counting 20 from the verbatim), so it is 80x5=300 more lines in the tor.onion-service.sh, also great part of the code is duplicated on those scripts, will pass them to torbox.lib or tor.onion-service.sh. I will remove instance and vanguards option from that script as it is not needed there for now. Total estimate is 600 lines in the tor.onion-service.sh. I am trying to keep it very short, no duplicate code there, functions and loops to refer too.

nyxnor, 2021-08-11:
Current problem I need help to bypass.
When selecting the option client auth from the menu, user can select multiple options. When that happens, I want to show the interjection of the authorization files with the same name that are present on differente services dir.
```
ssh/authorized_clients/alice.auth
ssh/authorized_clients/bob.auth

test/authorized_clients/alice.auth
```
If selecting ssh and test service, I want to display only alice.

radio24, 2021-08-11:
Taking the official documentation about setting up an Onion Service as a reference, are these files necessary for the Client Authorization? If I understand it right, with it, the owner of the server can decide who can access it and who not. How is it with a public server - let's say torbox.ch on an onion address? Does such a server need these .auth files too?

nyxnor, 2021-08-11:
Yes, those files are necessary for servers with client auth. [2019 guide](https://2019.www.torproject.org/docs/tor-manual-dev.html.en#_client_authorization) and the [clean community guide](https://community.torproject.org/onion-services/advanced/client-auth/).

The nomenclatures I use for clarification:
Hidden server - just known to user or to select group that was shared
Public server - normally used with onion location, to be known and seen by anyone

Hidden servers are appropriate to set up Client Authorization to manage who can view the service and access it, without the client key, the service page does not even load, protecting the privacy and against DOS attacks.

torbox.ch is supposed to be seen and used by anyone, so it should not require a client key. I know it is hard to grasp without testing, but I did a lot of improvements to the script and will try to release them soon after enough testing.

More about client auth:
The script I made automatically saves to the server authorized clients the pub key of each client, it echos the private key in the right format to be used in the torbrowser or to be saved to onion_auth/alice.auth_private, this is manual step by the client, you just need to pass them the key and directions, which will be done with infomation displayed with text/onion-auth-explanation-save-client-key

nyxnor, 2021-08-12:
Take a look here https://github.com/nyxnor/CLI-onion-services/tree/main
The checklist option for whiptail is too ugly, it is using that but with dialog checklist commented if needed to change.

nyxnor, 2021-08-13:
The cleanest guide to understadn client auth: https://matt.traudt.xyz/posts/creating-private-v3-FgbdRTFr/

nyxnor, 2021-08-24:
Updated repo https://github.com/nyxnor/onion-cli
It is very simple, the .md files are there cause I am planning on adding them to instructional part of the menu.
Onion-Location is just a guide.
Backup functionality such as export and import are not finished yet and need this to be complete (but not only this, constant bugs trying to solve).

nyxnor, 2021-09-21:
## CLI

https://user-images.githubusercontent.com/69700936/134138134-c054886d-b921-4d88-ae94-bf9f7795bebb.mp4

-------------------

## TUI

Overall view of the TUI, it just calls the cli script.
Also if you notice, on the TUI, it only appears more options that required a onion service if you have at least one configured.
lib + tui + cli = 1500 lines of code, which is short to review.

https://user-images.githubusercontent.com/69700936/134151696-7aeea39c-def8-474d-8197-037e89024ab1.mp4

connected201, 2021-09-22:
very nice, it will be implement in next release of torbox?

radio24, 2021-09-22:
Regarding the implementation, my idea is to do it in several steps:

1. Implementation of a public website (for example torbox.ch, which doesn't require any client authentication).
2. Implementation of private websites od "shares", which requires a client authentication.

**How about the time table**
To-do for the next 2-3 weeks:
- [ ] We will first fix some bugs in the current configuration, like the one described in issue #88 .
- [ ] We will also explorer the possibility to deactivate SSH as described in issue #83 and #84 .
- [ ] Some of the installed python modules and compiled packages (for example snowflake) are not automatically updated with the Update and Reset sub-menu - we will try to change that.

After completing the above points, we will start with the implementation of the hidden services.

nyxnor, 2021-09-23:
Can you make a github repo of the torbox.ch website so I can test publishing it? I see it is wordpress and they use cloudflare (`curl --head https://torbox.ch`).

Note that publishing a website is just one of the utilities one can do with onion domain. I use for instant messaging communication with XMPP an onion domain.

Note that implementing a website with or without client auth won't differ much, few clicks with the menu or some arguments with the CLI, which is way fast than doing manually.

My timetable is difficult to estimate, 
- [ ] I have being trying to fix all bugs that could cause incorrect torrc configuration (critical cause they make tor fail and need manual edit if occurs), then after it is stable (my script tries to stop any incorrect configuration before inserting into torrc. I can't skip this step, it is necessary.
- [ ] Error handling on `auth server list` when there is not client in there. This is not an error per se, it just does not output anything meaningful saying there is not client, and I want that to be very clear to the user.
-  [ ] Nginx configuration sample that don[t require user configuration, the script should handle inserting the tcp or unix path for the `listen` parameter. I havent started this yet. (Apache maybe if someone send me a apache configuration sample). Nginx resources from [MTigas ProPublica with Unix socket and a Single config for Tor and plainnet](https://gist.github.com/mtigas/9a7425dfdacda15790b2) and [Raspiblitz TCP socket with Separate config for Tor and local](https://github.com/rootzoll/raspiblitz/tree/v1.7/home.admin/assets/nginx)

nyxnor, 2021-09-23:
If one just want to publish a website via terminal, there is [onionshare-cli](https://github.com/onionshare/onionshare/tree/develop/cli), I like what they do but I am not sure it fits TorBox because it saves all information to a single chosen file, so they do not modify local torrc and HiddenServiceDir. Anyway, onionshare-cli is really helpful for setting a website to chat, send and receive files, publish a webpage and can add client auth on top of each previous action.

radio24, 2021-09-23:
> Can you make a github repo of the torbox.ch website so I can test publishing it? I see it is wordpress and they use cloudflare (`curl --head https://torbox.ch`).

Not at the moment. Just use a "Hello world" index.html, the [README.md ](https://github.com/radio24/TorBox/blob/master/README.md) or the static index.htm from https://torbox.ch for testing. At the end, the of a publication of a public website shouldn't orient it on https://www.torbox.ch, but just work for a usual html/php website.

> Note that publishing a website is just one of the utilities one can do with onion domain. I use for instant messaging communication with XMPP an onion domain.
That sounds interesting and could be another longterm project. OnionShare is a good example what is possible.

nyxnor, 2021-09-25:
Ok, got it working with one virtual port, the 80. For virtual port 443 with https on onion, will implement later.

1. Save the website files inside `/var/www/`. As an example, the folder will be called `torbox.ch`. The service will also be called `torbox.ch` for ease to remember.
```sh
sudo mkdir -p /var/www/torbox.ch
```

2. Create the onion service with `unix` or `tcp` socket:
SYNTAX: *on SOCKET SERVICE VIRTPORT*
```sh
sh onionservice-cli on unix torbox.ch 80
```

3. Activate the web server (nginx in this case):
SYNTAX: *WEBSERVER SERVICE FOLDER_INSIDE_VAR_WWW*
```sh
sh onionservice-cli nginx torbox.ch torbox.ch
```

Done, that is all you need to activate an onion service and spin up the web server.
The menu option will come after I manage to make it work with the virtual port 443.

nyxnor, 2021-09-26:
Ok, now it is available with nginx or apache2, you can only choose one of course.
Also, menu option to activate or deactivate serving a web site (not the hidden service configuration, the web server configuration for that hidden service)
set the web server in `.onionrc`:
[image]
Setup the environment with `sh setup.sh`
Call from any folder the `onionservice-tui`:
[image]
[image]
[image]
[image]
[image]
[image]

-----

pardon my html/css page, I am still learning how to do it.

radio24, 2021-09-26:
That looks really great!

nyxnor, 2021-09-26:
But now you want just a you and select clients to access that page, add client auth on your server:
[image]

Lets generate a key pair a this is an example, but if the client gave you only his pub key, select the option below of course:
[image]

[image]

Lets add alice and bob, possible combinations for clients are: `alice bob`, `alice,bob`, alice, bob`. Comma or space separated.

[image]

Important line for Tor Browser is `CLIENT_PRIV_KEY=KABLUFI3AMAY62OPOZL3PFHXKMFRE5XCBXBDVLHHAA5V7RNEWZWA`
Important line for tor (daemon) is
`CLIENT_PRIV_KEY_CONFIG=sjyyyi6y3hqszp4myx3ytghdjpe6gavldymkbwj62wvhsxxs2lg7njqd:descriptor:x25519:KABLUFI3AMAY62OPOZL3PFHXKMFRE5XCBXBDVLHHAA5V7RNEWZWA`

Lets copy the browser key option. Reload the Browser page and try again
[image]

Now it is authenticated, you need a key to access it.

Try leaving the key field blank or with wrong credentials:
[image]

radio24, 2021-09-26:
Unfortunately, it will take me a little longer to integrate it into TorBox. My job (the one that pays my bills) is putting me under a bit of workload again.

nyxnor, 2021-09-26:
No problem, with that I have more time to test and to find a job :)
The warning on the readme to not trust the repo is because it was only reviewed by myself and developers can be blind on the code they wrote. Also the backup technique needs to be improved, it does its job but I will always want a better backup to be safe.

About you integrating into TorBox, note that you just need to clone and follow the steps on the readme, maybe add one option on the menu to redirect to onion services menu. There is no secret.

But if you really decide to go deep and integrate each option your way, that is okay but I tried to integrate TorBox to Raspiblitz and it worked the first time, but with each new commit and release of TorBox, it becomes harder to keep upgrading individually the files I've changed. The same goes for integrating OnionService to TorBox. (The end result is that I am still building a custom script to change hardcoded paths of TorBox with sed and try to not break things when bulking with sed, which is difficult when dealing with multiple files).

nyxnor, 2021-09-26:
About the long information first comment https://github.com/radio24/TorBox/issues/80#issue-958601443

I have completed all it is possible to do scripting for an onion service. The rest is OpSec that depends on the operator such as:
> different machine: consider running the onion service on a different machine (real or virtual) than the actual service. This has the advantage that you can isolate the service from the onion service (a compromise of one doesn’t compromise the other) and helps with isolating potential information leaks
isolation: similarly to the above, you can also isolate Tor and the service so it will run on a different network namespace than the service. Tails uses a Tor-or-fail packet filter.

OpSec recommendation to use unix domain as the HiddenServicePort VIRTPORT TARGET is done already for example.

radio24, 2021-09-26:
I didn't decide yet on which approach I will integrate onionservice-cli and onionservice-tui. At first sight, it seems that I can take onionservice-cli as it is but have to (re-)write my version of onionservice-tui. 

One reason I will not only copy your work without understanding the code is that I will also try to contribute to your work. I already did a fork of your project, and if I find something I can contribute, I will make a pull request.

Anyway, great work!

nyxnor, 2021-09-27:
Feel honored, thanks for the fork. Having any questions about `why` I did something just say it, because the documentation is more about what the `result` of a combination of command will be, instead of `why` and `how` it is doing that.

Also the TUI problem is the color (that you can set with .dialogrc) or because I use `dialog` instead of `whiptail`?
I am asking this because I tried to make the most configurable possible so anyone that wants to use it won't have trouble modifying the scripts, if you explain what is lacking I can do enhance.

radio24, 2021-09-27:
What are the advantages of `dialog` over `whiptail`?

nyxnor, 2021-09-27:
[FreeBSD's dialog man (read whiptail section in there also)](https://www.freebsd.org/cgi/man.cgi?query=dialog&sektion=1)

* Whiptail is dialog with a less boxes options
* with dialog I can move between options with the first letter, so it is faster to navigate than just arrow keys
* dialog centers lists within the window. whiptail  typically puts lists against the left margin. I think centering is better.
* I did not like that whiptail highlights in red stopping at the last character, with dialog, it highlights the whole line
* color choice, does not need to be dark, can be green for TorBox editing .dialogrc.
* I just discovered there is `dselect` and `fselect` with dialog, which I can delect directories and files.

nyxnor, 2021-10-05:
green dialogrc to fit torbox

[image]

```
# Run-time configuration file for dialog
#
# Types of values:
#
# Number     -  <number>
# String     -  "string"
# Boolean    -  <ON|OFF>
# Attribute  -  (foreground,background,highlight?,underline?,reverse?)
#
#
# Colors:
#
# Interpret embedded "\Z" sequences in the dialog text by the following character,
# which tells dialog to set colors or video attributes: 0 through 7 are the ANSI
# used in curses: black, red, green, yellow, blue, magenta, cyan and white
# respectively. Bold is set by 'b', reset by 'B'. Reverse is set by 'r', reset by 'R'.
# Underline is set by 'u', reset by 'U'. The settings are cumulative, e.g., "\Zb\Z1"
# makes the following text bold (perhaps bright) red. Restore normal settings with "\Zn".
#
# 0 = black
# 1 = red
# 2 = green
# 3 = yellow
# 4 = blue
# 5 = magenta
# 6 = cyan
# 7 = white
#
# b = bold (set)
# B = bold (unset)
# r = reverse (set)
# R = reverse (unset)
# u = underline (set)
# U = underline (unset)
# n = default (restore)
#
# Source: https://github.com/openoms/joininbox/blob/v0.6.0/scripts/.dialogrc

# Set aspect-ration.
aspect = 0

# Set separator (for multiple widgets output).
separate_widget = ""

# Set tab-length (for textbox tab-conversion).
tab_len = 0

# Make tab-traversal for checklist, etc., include the list.
visit_items = OFF

# Shadow dialog boxes? This also turns on color.
use_shadow = OFF

# Turn color support ON or OFF
use_colors = ON

# Screen color
screen_color = (WHITE,GREEN,ON)

# Shadow color
shadow_color = (BLACK,BLACK,ON)

# Dialog box color
dialog_color = (BLACK,WHITE,ON)

# Dialog box title color
title_color = (BLACK,WHITE,ON)

# Dialog box border color
border_color = (BLACK,WHITE,ON)

# Active button color
button_active_color = (WHITE,GREEN,ON)

# Inactive button color
button_inactive_color = dialog_color

# Active button key color
button_key_active_color = button_active_color

# Inactive button key color
button_key_inactive_color = (GREEN,WHITE,OFF)

# Active button label color
button_label_active_color = (WHITE,GREEN,ON)

# Inactive button label color
button_label_inactive_color = (GREEN,WHITE,ON)

# Input box color
inputbox_color = dialog_color

# Input box border color
inputbox_border_color = dialog_color

# Search box color
searchbox_color = dialog_color

# Search box title color
searchbox_title_color = title_color

# Search box border color
searchbox_border_color = border_color

# File position indicator color
position_indicator_color = title_color

# Menu box color
menubox_color = dialog_color

# Menu box border color
menubox_border_color = border_color

# Item color
item_color = dialog_color

# Selected item color
item_selected_color = button_active_color

# Tag color
tag_color = title_color

# Selected tag color
tag_selected_color = button_label_active_color

# Tag key color
tag_key_color = button_key_inactive_color

# Selected tag key color
tag_key_selected_color = (RED,GREEN,ON)

# Check box color
check_color = dialog_color

# Selected check box color
check_selected_color = button_active_color

# Up arrow color
uarrow_color = (GREEN,BLACK,ON)

# Down arrow color
darrow_color = uarrow_color

# Item help-text color
itemhelp_color = (BLACK,WHITE,OFF)

# Active form text color
form_active_text_color = button_active_color

# Form text color
form_text_color = (BLACK,WHITE,ON)

# Readonly form item color
form_item_readonly_color = (GREEN,BLACK,ON)

# Dialog box gauge color
gauge_color = title_color

# Dialog box border2 color
border2_color = dialog_color

# Input box border2 color
inputbox_border2_color = dialog_color

# Search box border2 color
searchbox_border2_color = dialog_color

# Menu box border2 color
menubox_border2_color = dialog_color
```

radio24, 2021-10-16:
I studied the code of the "onionservice" scripts, and I'm starting to implement some of the features in TorBox v.0.4.3.

"onionservice" gives users the possibility to use "tcp socket" or "UNIX socket". Because "UNIX socket" is the more secure way, I ask myself if I should use this selection as default in TorBox. Are there some disadvantages if "UNIX socket" is used instead of "tcp socket"?

nyxnor, 2021-10-17:
I have nothing to complain about unix socket.
<details>
<summary>There is an old bug with nginx unix socket but was fixed. Just drop below if interested. </summary>
<br>
There was a problem of it not being removed when [using SIGQUIT for the nginx process](https://trac.nginx.org/nginx/ticket/952), but it was fixed 17 months ago according to [this commit](https://trac.nginx.org/nginx/changeset/7cbf6389194b9170514e514ca7ee495369c9c8ac/nginx). I am saying this because this issue stood unfixed for more than 6 years since the issue was opened and people still refer to it for not using unix socket and not search it was closed, maybe trauma.
</details>

nyxnor, 2021-10-17:
git pull my main branch, I fixed a bug that was implemented here https://github.com/radio24/onionservice/commit/85f95ba6961c01eeb9a1352dc698947c2ac9f20e, fix here https://github.com/nyxnor/onionservice/commit/3958e6d603613fdfe374bad53e3e6c45ad0b61cc

radio24, 2021-11-24:
@nyxnor How should we treat this thread?
- Keep it open under issue?
- Close it?
- Move it to discussions (archive)?

nyxnor, 2021-11-24:
discussions, there will be always something new to add and other people to report their experiences

closed 2021-11-24 by radio24