zotil/TorBox #81: Please correct download link from github release page!

connected201, 2021-08-05:
Correct link from torbox.ch        `https://www.torbox.ch/data/torbox-20210802-v042.gz`
Broken link from github release page   `https://www.torbox.ch/data/torbox-20210208-v042.gz`

radio24, 2021-08-05:
Thank you for noticing - it is now fixed.
If you didn't download it yet, I would recommend waiting until tomorrow evening. I will put a new image (with the same name) online, with some bugs fixed. After its release, I will put a note in here in this thread.

### Fixed bugs with the coming up image update:

- **BUG**: The current image file is built with [Linux Kernel 5.10.52](https://kernelnewbies.org/Linux_5.10). Unfortunately, we didn’t realize that all the additional network drivers (Realtek 8188eu, 8188fu, 8192eu, 8812au, 8812bu, 8814au, 8821au, 8821cu, and 8822bu) are not yet available for this Linux Kernel version. The latest supported Linux Kernel is version 5.10.49. Those already working with the current image file or who have updated the system and need one of these network drivers can fix the bug by updating the TorBox menu ([update and reset sub-menu entry 5](https://www.torbox.ch/?page_id=1410)) and executing the command below on TorBox’s command prompt. Afterwards, rerun menu entry 2 in the [Update and Reset sub-menu](https://www.torbox.ch/?page_id=1410). We will fix the image by the end of the week by switching back to the [Linux Kernel 5.10.49](https://kernelnewbies.org/Linux_5.10) and installing the network drivers. **BUG FIXED✔︎ – IMAGE FILE FIXED SOON!** ⏳
```
cd ~
sudo bash install/step_back_to_kernel_5.10.49
```

- **BUG**: The integration of TorBox’s automatic counteractions into `rc.local` resulted in some nasty bugs, which we didn’t see before: `rc.local` needs absolute paths, not relative ones; there was also an error in getting the name of the internet interface from the run-file, which broke the execution of the `automat` script. We also put the logs of TorBox’s automatic counteractions into a separate file (`/var/log/tor/automat.log`) so that it is easier to see if automatic counteractions were activated or not. You can fix the bug by updating the TorBox menu ([update and reset sub-menu entry 5](https://www.torbox.ch/?page_id=1410)). The image will be fixed by the end of the week. **BUG FIXED✔︎ – IMAGE FILE FIXED SOON!** ⏳

- **BUG**: Due to a little bug in the script, TorBox tells during the activation of the Meek-Azure bridge that the bridge is offline, which is not the case. However, you can continue, and the bridge will work without any problem. You can fix the bug by updating the TorBox menu ([update and reset sub-menu entry 5](https://www.torbox.ch/?page_id=1410)). The image will be fixed by the end of the week. **BUG FIXED✔︎ – IMAGE FILE FIXED SOON!** ⏳

- **BUG**: Pressing the enter key in the OBFS4 port definition (“port number of the OBFS4”) during the OBFS4 Bridge Relay configuration will not automatically take the latest used or the default number. This will result to a faulty `torrc` entry (`ServerTransportListenAddr obfs4 0.0.0.0:` instead of, for example, `ServerTransportListenAddr obfs4 0.0.0.0:443`). This will prevent tor from starting until the S`erverTransportListenAddr` line is fixed or deactivated. The workaround is easy: don’t press the enter key during the OBFS4 Bridge Relay configuration, but write the number into the dialogue. However, you can fix the bug by updating the TorBox menu ([update and reset sub-menu entry 5](https://www.torbox.ch/?page_id=1410)). The image will be fixed by the end of the week. **BUG FIXED✔︎ – IMAGE FILE FIXED SOON!** ⏳ ⏳

connected201, 2021-08-05:
Question, why i cannot access torbox under local network? my torbox local ip 192.168.1.35.
SSH port not opened from LAN? if yes, how i can enable access to torbox truth LAN?

radio24, 2021-08-05:
There are two interfaces, one supposed to be connected with the Internet (in your case, probably the one with the IP 192.168.1.35) and another for client's access (192.168.42.1 or 192.168.43.1). For security reasons, the SSH access from the Internet side is by default blocked. However, you can open that access with entry 10 in the [Configuration sub-menu](https://www.torbox.ch/?page_id=875).

See also [issue #26](https://github.com/radio24/TorBox/issues/26) and [issue #46](https://github.com/radio24/TorBox/issues/46).

connected201, 2021-08-05:
thanks, all working fine.

radio24, 2021-08-05:
I will close the issue as soon as the revised image file is online.

radio24, 2021-08-06:
The new image (with the same name) is now online and fixes the bugs listed above.

closed 2021-08-06 by radio24