zotil/TorBox #82: how to correctly import and export hidden services?

connected201, 2021-08-06:
Hi, i saved my hidden services from previously version of torbox, /var/lib/tor and torrc file, but after copy folder with hidden services and add hidden service dir records to torrc, torbox don't want to share internet, in finally internet not working anymore, what i do wrong?

radio24, 2021-08-06:
Until yet, TorBox doesn't support hidden services.

connected201, 2021-08-06:
> Until yet, TorBox doesn't support hidden services.

you mean that i will make records in torrc like:
```
HiddenServiceDir /var/lib/tor/hidden_service/ 
HiddenServicePort 80 127.0.0.1:80
```
hidden service will not work? because i do this, and hidden service worked, but, i want to import my hidden services, all devices connected to torbox lost internet connection after `cp -R tor /var/lib/tor/ `with my hidden services.

radio24, 2021-08-06:
No, I mean that I didn't look into Hidden Services yet.

Probably, with your command, you created `/var/lib/tor/tor`. I don't think that's right and because of`HiddenServiceDir /var/lib/tor/hidden_service/` in `torrc` that could block tor from starting up. You may try `cp -R hidden_service /var/lib/tor/`. Another problem could be that the owner of the directory hidden_service and all its folders/files are wrong. It should be `debian-tor:debian-tor`. You may try the following commands after copying the folder:
```
chown -R debian-tor:debian-tor /var/lib/tor/hidden-service/
chmod -R 0700 /var/lib/tor/hidden_service/
```

connected201, 2021-08-07:
worked, thanks

closed 2021-08-07 by connected201