zotil/TorBox #83: how to disable SSH access torbox from LAN and WLAN?

connected201, 2021-08-06:
Hi, how to disable possibility to connect on torbox from devices connected from 192.168.43.x and 192.168.42.x ?

radio24, 2021-08-06:
Currently, if you are connected with a cable, you can only disable the TorBox Wifi, which disables `192.168.42.x`
See [entry 9 in the Configuration menu](https://www.torbox.ch/?page_id=875).

If you want to disable **the SSH access** from the client side (`192.168.43.x` and `192.168.42.x`), probably the easiest way is to deactivate/remove the following line in set_interfaces:
```
$IPTABLES -t nat -A PREROUTING -i $I_DEVICE -d $MY_OWN_IP -p tcp -j REDIRECT
```

..and in set_interfaces_2:
```
$IPTABLES -t nat -A PREROUTING -i $I_DEVICE1 -d $MY_OWN_IP1 -p tcp -j REDIRECT
$IPTABLES -t nat -A PREROUTING -i $I_DEVICE2 -d $MY_OWN_IP2 -p tcp -j REDIRECT
```

connected201, 2021-08-07:
sorry, but how i can remove these lines?
```
torbox@TorBox042:~/torbox $ IPTABLES -t nat -A PREROUTING -i $I_DEVICE -d $MY_OWN_IP -p tcp -j REDIRECT
-bash: IPTABLES: command not found
```

connected201, 2021-08-07:
i'm not familiarized with iptables, :(

radio24, 2021-08-08:
If I were you, I would test the change first by changing the active iptables:
`sudo iptables -t nat -D REDIRECT -p tcp -d 192.168.42.1`

That should prevent the possibility to make a new SSH connection to the TorBox. If the outcome is what you were looking for, you can make it permanent by opening `set_interfaces` and `set_interfaces_2` (in `/home/torbox/torbox` type `nano set_interfaces` / `nano set_interfaces_2` and removing the three lines mentioning above.

connected201, 2021-08-08:
`torbox@TorBox042:~/torbox $  sudo iptables -t nat -D REDIRECT -p tcp -d 192.168.42.1
iptables v1.8.2 (nf_tables): Chain 'REDIRECT' does not exist
`

what i do wrong?

radio24, 2021-08-08:
Ah, sorry, it is my error. The chain is called PREROUTING and the command is the following:
`sudo iptables -t nat -D PREROUTING -p tcp -d 192.168.42.1`

connected201, 2021-08-08:
> 
> 
> Ah, sorry, it is my error. The chain is called PREROUTING and the command is the following:
> `sudo iptables -t nat -D PREROUTING -p tcp -d 192.168.42.1`

torbox@TorBox042:~/torbox $  sudo iptables -t nat -D PREROUTING -p tcp -d 192.168.42.1
iptables: Bad rule (does a matching rule exist in that chain?).

radio24, 2021-08-08:
Could you give me the output of `sudo iptables -t nat -L`?

connected201, 2021-08-08:
```
torbox@TorBox042:~/torbox $ sudo iptables -t nat -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
REDIRECT   tcp  --  anywhere             192.168.42.1
REDIRECT   tcp  --  anywhere             anywhere             redir ports 9040
REDIRECT   udp  --  anywhere             anywhere             udp dpt:domain redir ports 9053
REDIRECT   udp  --  anywhere             anywhere             redir ports 9040

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination
MASQUERADE  all  --  anywhere             anywhere

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
```

radio24, 2021-08-08:
Try the following to remove line number 1 (`REDIRECT   tcp  --  anywhere             192.168.42.1`):
`sudo iptables -D PREROUTING 1`

connected201, 2021-08-08:
torbox@TorBox042:~/torbox $  `sudo iptables -D PREROUTING 1`
iptables v1.8.2 (nf_tables): Chain 'PREROUTING' does not exist

radio24, 2021-08-08:
How about: `sudo iptables -t nat -D PREROUTING 1`

connected201, 2021-08-08:
> 
> 
> How about: `sudo iptables -t nat -D PREROUTING 1`

executed succesfully!

connected201, 2021-08-08:
```

torbox@TorBox042:~/torbox $ sudo iptables -t nat -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
REDIRECT   tcp  --  anywhere             anywhere             redir ports 9040
REDIRECT   udp  --  anywhere             anywhere             udp dpt:domain redir ports 9053
REDIRECT   udp  --  anywhere             anywhere             redir ports 9040

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination
MASQUERADE  all  --  anywhere             anywhere

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

```

radio24, 2021-08-08:
> what next? i want to block access from LAN 192.168.43.x
Try to make a new ssh connection. That should fail.

connected201, 2021-08-08:
all work fine, also i saved iptables with `netfilter-persistent save`

connected201, 2021-08-08:
maybe you add in next release option to disable SSH from `192.168.42.x `and `192.168.43.x` ? it will be very useful. thanks for active work of this project.

radio24, 2021-08-08:
The iptable configuration will only be persistent, if you make the changes into `set_interfaces` and `set_interfaces_2` as described above.

> maybe you add in next release option to disable SSH from 192.168.42.x and 192.168.43.x ? it will be very useful. thanks for active work on this project.

Currently, you were the only one requesting a such feature. I will think about adding it as a permanent feature to TorBox if more people request it.

connected201, 2021-08-08:
i installed l` iptables-persistent`, because for me it's easiest way to run `sudo iptables -t nat -D PREROUTING 1 && netfilter-persistent save` 
also, as i understand command `iptables -t nat -D PREROUTING 1` work for both network interfaces?

radio24, 2021-08-08:
Yes, but every time you change something on the network configuration (main menu 5-10), the saved iptables rules will be overwritten.

radio24, 2021-08-10:
Connected with issue #84.

radio24, 2021-08-10:
I reopened the issue again, so that I don't forget to look into it with the next TorBox version.

radio24, 2025-07-13:
Ok, the user with the Issue / Request had been deleted in the meantime. Nevertheless, I will finally add SSH access restrictions to the Danger Zone for the v.0.5.5 release and have just started working on it. The tasks are:
- [ ] Moving all SSH-related menu entries from other sub-menus (in particular from the Configuration sub-menu) to the Danger Zone.
- [ ] Creating a selection menu, where I can select if I want to deactivate/activate SSH completely and alternatively can select which connection should be allowed and which not (from the Internet, cable or WiFi).
- [ ] Checking if WebSSH also needs to be modified.
- [ ] Enable Tor control port access for clients (in the Configuration sub-menu) can be moved to the Tor section.
- [ ] Disable HTTP plain text traffic block (in the Configuration sub-menu) can be moved to the Network section.
- [ ] Checking where the entry "Enable root access" should be moved (Danger Zone)?

closed 2025-07-17 by radio24