zotil/TorBox #84: Automatic enable access point if no internet

connected201, 2021-08-09:
Hi, very nice will be, if torbox automatically enable access point (TorBox042) if torbox don't have internet, to access torbox and connect to mobile wifi on example, or another network source, otherwise i cannot access torbox, and set up, only with eth0, in some case it's difficult or impossible to do this.

radio24, 2021-08-10:
I agree with you. However, you can also connect a monitor and keyboard to the TorBox and log in.

I guess, you locked yourself out with the option to disable the access point (entry 9 in the Config sub-menu) or the option to disable the access on TorBox itself from outside as you did with [issue 83](https://github.com/radio24/TorBox/issues/83) (which is not implemented yet).

connected201, 2021-08-10:
i enabled access to torbox from internet, so i can access it from internet if i connected in same subnet where is connected torbox, but i want to have portable hotspot with battery, i need possibility to play with torbox settings, i want to make a simple web interface in node red dashboard, to scan and connect to my wifi via nmcli.

closed 2023-01-01 by radio24