zotil/TorBox #85: Get No Internet Connection at all

SonicMetallica, 2021-08-14:
Hello i use a Lan Cable to connect my Raspberry Torbox to the normal Router and i use my WLAN Adapter plugged into my PC to connect to the Torbox Onboard WLAN. So i should have internet?

Or do i have to use with this hardware either a additional Wlan-Adapter or a USB-Lan Adapter?

I tried every of the option for Internet connection which could possible be. But i dont get it work. I  am really desperate please help me. After 2 Day trying everything out i possible come up with and using the sufu i am done with my proud.

radio24, 2021-08-14:
I'm sorry to hear about your problems with TorBox.

Your configuration should work without problem. Although, it could be that with the first start after the question, if you want to use OBFS4 bridges that TorBox will not connect right away. In this case, press CTR-C to leave the tor bootstrapping log and choose entry 5 in the [main menu](https://www.torbox.ch/?page_id=775). With entry 4, you should see tor bootstrapping until it reaches 100%. The data traffic of the clients connected to the TorBox042 wifi will be routed through tor.

If tor is not bootstrapping with main menu entry 5, it could be that TorBox doesn't receive an IP address from your Internet router/modem. You can check that by using the command ifconfig on TorBox's command-line interface (just leave the Torbox main menu with ESC). For more information see [here](https://www.torbox.ch/?page_id=112#my-torbox-doesnt-receive-an-ip-address-from-the-network-router) or [here](https://www.torbox.ch/?page_id=112#my-torbox-receives-an-ip-address-19216842-or-19216843-from-the-network-router-but-it-doesnt-work).

If this doesn't help, [check out this FAQ entry](https://www.torbox.ch/?page_id=112#i-cant-get-torbox-to-work-and-need-more-detailed-help-what-should-i-do), what you should [send us in an email](mailto:anonym@torbox.ch) so that we can try to help you.

SonicMetallica, 2021-08-14:
After i did change to captive with option 5 i had for a short time internet then i tried to update and it was gone. After the failed update i did the same but didnt get any connection.

radio24, 2021-08-14:
This is not helping me to solve your problem. The log is too short - it can take up to 5 minutes to get a tor connection. Sometimes, when tor is stuck, [entry 9 in the Update and Reset sub-menu](https://www.torbox.ch/?page_id=1410) helps. Also, the IP of your eth0 is crucial. For example, if it is some kind of 169.x.x.x, most probably it is a self-assigned IP, which indicates a problem with the DHCP of your router/modem.
There is no need to post your configuration in that thread. Again, check out [the FAQ entry](https://www.torbox.ch/?page_id=112#i-cant-get-torbox-to-work-and-need-more-detailed-help-what-should-i-do), what you should send us [in an email](mailto:anonym@torbox.ch) so that we can try to help you.

connected201, 2021-08-17:
make again clean install, and will work, i have one time same trouble, when i start OBFS4, or when first setup starts, and torbox kick me from ssh, i don't remember exactly.

radio24, 2021-08-17:
If you don't need OBFS4 bridges, try without them. Using OBFS4 bridges with a tight firewall configuration on the router can lead to issues.

Papiloco93, 2021-08-29:
Hello, i have the same problem like SonicMetallica. Do you have a Solution for these problem?
its the second day, I try to fix the problem.....

radio24, 2021-08-29:
Yes, see here: [I selected to use OBFS4 bridges with pluggable transports when I started TorBox the first time, but so far I’m unable to connect to the Internet. TorBox received a valid IP from the router, but tor stuck at “Bootstrapped 0%”. What can I do?](https://www.torbox.ch/?page_id=112#i-selected-to-use-obfs4-bridges-with-pluggable-transports-when-i-started-torbox-the-first-time-but-so-far-im-unable-to-connect-to-the-internet-torbox-received-a-valid-ip-from-the-router-but-tor-stuck-at-bootstrapped-0-what-can-i-do)

closed 2021-09-04 by radio24