zotil/TorBox #88: torbox lost connection with my wifi,

connected201, 2021-09-14:
hi, if i set "where is the internet" from wlan0, after all settings all work fine, but after reboot raspberry, torbox lost connection with my wifi and appear as acces point "TorBox042", i disabled from menu to don't create access point, but torbox create access point every time after reboot, and cannot automatically connect to my wifi, how to fix it?

radio24, 2021-09-18:
Finally, I had time to look at the bug. If you are only interested in the solution, skip to "How to fix the bug".

**Explanation**
In a default installation with wifi activated, a fall-back routine in `rc.local` checks if `wlan1` is configured as "static", which would mean that `wlan1` would be used as AP (TorBox's WLAN). However, to have `wlan1` as TorBox's WLAN after a restart is risky because before or during TorBox's downtime, someone could pull the USB wifi dongle (wlan1), which most probably would lead to a lock-out situation.

**Bug**
If TorBox's WLAN is disabled, this fall-back routine triggers to re-activate it. We thought that `sudo systemctl disable hostapd` would be sufficient that the fall-back routine will have no effect. This assumption was wrong. However, this is only the case if the user takes the internet from wlan0.

**Possible solutions**
1. Possibility to use separate `interfaces` files if TorBox's WLAN is disabled. -> this will be complicated in the long term.
2. Possibility to use `sudo systemctl mask --now hostapd`, which is stronger than `sudo systemctl disable hostapd` -> good 
solution, but `rc.local` shouldn't execute code which we know that it is not working or conflicting with other parts of TorBox.
3. Possibility to use separate `rc.local` files if TorBox's WLAN is disabled. -> this is much easier and controllable than option 1.

We decided to implement options 2 and 3.

**Side effects & linked issues**
When investigating the bug, we also found another issue. Switching from wifi to ethernet is too complicated. If someone suddenly uses a cable connection to the TorBox without configuring it properly (which is not very good explained in the documentation), a restart will not help. The intended behaviour should be that a user always gets the TorBox menu irrespective of the kind of connection to the TorBox. For historical reasons, the problem here is that we handle wifi different from cable connections, reflected in two other scripts (`set_interfaces` and `set_interfaces_2`). Also, the TorBox's wifi IP address was statically defined into `set_interfaces`, which explains the problem people were faced when using only cable connection without wifi. The bug-fix will also address this issue and switch all corresponding code portions to `set_interfaces_2` and replace `set_captive` with `set_captive_2`.

**How to fix the bug**
TorBox's master branch is updated with the fix. To implement the bug-fix on the local TorBox, use [entry 5 in the Update and Reset sub-menu](https://www.torbox.ch/?page_id=1410).

connected201, 2021-09-19:
Hi, unfortunately problem not solved + i have some bug with wireless manager, wifi not accept my password, but password is correct, and also, for wireless network with open network, wireless manager ask password to connect to open network :)
 
[image]

also after reboot raspberry, raspberry connect to my AP but it do very strange, because ip in status have this view and internet in this time not working on raspberry, i cannot ping any host:
[image]
and if i quit from wireless manager and enter again in wireless manager, i receive real ip and i can make ping and internet working properly.
[image]

radio24, 2021-09-19:
Let's fix one problem after another. What does "problem not solved" mean in relation to your first entry? Does TorBox still appear as "TorBox042" after a restart? If this is the case: did you enable and disable again TorBox's WLAN in the Configuration sub-menu to be sure that the changes take effect?

connected201, 2021-09-19:
hi, problem not resolved:
Q: Does TorBox still appear as "TorBox042" after a restart? If this is the case: did you enable and disable again TorBox's WLAN in the Configuration sub-menu to be sure that the changes take effect?
A: TorBox042 don't appear after configuration and disabling torbox WLAN permanently.
Q: What does "problem not solved" mean in relation to your first entry?
A: After restart raspberry don't connect to my wireless network automatically! i need to enter in menu: 7. Wireless network (trough the onboard chip; wlan 0) and wait when wireless manager will scan all network, and after this, in left down corner i see status with connected to wifi 
[image]
after quitting from wireless manager i have internet,

radio24, 2021-09-19:
Ok, let's try my additional fix, just pushed to the master branch. I changed `rc.local` again, which only initiated an automatic reconnection on wlan1 but not on wlan0.  

To implement the bug-fix on the local TorBox, use [entry 5 in the Update and Reset sub-menu](https://www.torbox.ch/?page_id=1410). After that enable and disable again TorBox's WLAN in the [Configuration sub-menu](https://www.torbox.ch/?page_id=875) to be sure that the changes take effect.

If you execute `sudo less /etc/rc.local` after applying the fix, you should see the following entries:
```
if ip link | grep wlan0 | grep "state DOWN" ; then
  sudo /usr/bin/python3 /home/torbox/torbox/torbox_wireless_manager.py -i wlan0 -a
  sudo sed -i "s/^INTERNET_IFACE=.*/INTERNET_IFACE=wlan0/" /home/torbox/torbox/run/torbox.run
  sleep 5
fi
```

connected201, 2021-09-19:
i will make clean install of torbox tomorrow, and we will test on fresh installed torbox

connected201, 2021-09-20:
on clean installed torbox, after connecting to my wifi i get wrong raspberry address (real ip address need to be like 192.168.1.125) :
[image]

question, what mean this logs:
[image]
[image]
also

radio24, 2021-09-20:
Let's put the auto-connection problem on the side at the moment. I'm returning to your second article, where you try to connect a wireless network called "5G" at the end. Under the column "Sec", it is written that this is an ESS, which is the abbreviation for an Extended Service Set (802.11s). An ESS is a wireless network created by multiple access points, which appears to users as a single, seamless network, such as a network covering a home or office that is too large for reliable coverage by a single access point. I have only WPA, WPA2 and WEP networks in my area but no ESS, which I can test. I will give that issue to the programmer of TorBox's Wireless Manager (TWM), Zotil. However, it would be helpful if you could provide us with some more information about the network.

1. Is this network build by specific hardware (company, brand, particular configuration)?
2. Is this network public without any password (maybe with a captive portal), or a password is necessary to connect to the network?
3. If I correctly understand your post, you have to connect the ESS network with TWM, close the TWM and connect the ESS network a second time, which shows you the correct IP address? Only after the second time, TorBox can use the ESS network?

To your question about the log entries: these entries shows that tor cannot connect a tor relay. Do you use OBFS4 bridges?

Zotil, 2021-09-21:
Connecting to an open network it should not ask for password. I will look into this issue.

@connected201 please provide some details to look in the restart issue.
- Does the open network have DHCP server enabled?
- Does the open network have MAC Filtering?
- Do you have any problem connecting to an encrypted network?
- The network that you expect to be connected after reboot, is the open network or encrypted?

Thank you

connected201, 2021-09-22:
Does the open network have DHCP server enabled? = YES
    Does the open network have MAC Filtering? = NO
    Do you have any problem connecting to an encrypted network? =YES, wireless manager ask me password again and again for my network with password, (but not all time, for this reason i open my 5Ghz wifi to connect without password, but some times wireless manager ask me password for open network to)
    The network that you expect to be connected after reboot, is the open network or encrypted? = OPEN

connected201, 2021-09-22:
on fresh installed torbox, i have same issue with password!, i can try to connect using nmcli, and we can see if raspberry ask password, or not! i want to repeat that password ask for open network and for my network, where i know my password, but correct password was ignored

connected201, 2021-09-22:
> Let's put the auto-connection problem on the side at the moment. I'm returning to your second article, where you try to connect a wireless network called "5G" at the end. Under the column "Sec", it [...]
Router Model: huawei echolife hg8145v5
>     2. Is this network public without any password (maybe with a captive portal), or a password is necessary to connect to the network?
No, my router share 2 wireless network 1. 2.4Ghz with password . 2. 5Ghz open network.
> 
>     3. If I correctly understand your post, you have to connect the ESS network with TWM, close the TWM and connect the ESS network a second time, which shows you the correct IP address? Only after the second time, TorBox can use the ESS network?
yes, when i have correct network IP, otherwise i have like mac address instead of ip address
> 
> 
> To your question about the log entries: these entries shows that tor cannot connect a tor relay. Do you use OBFS4 bridges?
no (i think)

connected201, 2021-09-26:
hi, problem with password, is fault for my router, (i solved this) but raspberry don't want to connect automatically to my wifi after restart

Zotil, 2021-09-29:
Hello @connected201 how did you solved it? It was a filter that you have on your router or some? Please let us know to understand better the issue.

I did tests with two different routers with and without password, I got the first connection without any problems. After reboot there is an issue but it will be fixed on the next version.

The error that show TWM after problems connecting to a free network is erroneous and it will be changed in the next version.

radio24, 2021-09-29:
> hi, problem with password, is fault for my router, (i solved this) but raspberry don't want to connect automatically to my wifi after restart

Let's focus now on the autoconnection of the wireless network after a reboot. I did the following:
- Updated my TorBox with [entry 5 in the Update and Reset sub-menu](https://www.torbox.ch/?page_id=1410) (necessary to get my latest fixes regarding the autoconnection feature on the startup).
- Connected the client with an ethernet cable to `eth0`.
- Went to the [Configuration sub-menu](https://www.torbox.ch/?page_id=875) and permanently disabled TorBox's WLAN (entry 9).
- Selected in the [main menu](https://www.torbox.ch/?page_id=775) `wlan0` as the source for the Internet (entry 7). The first time, this will take some time because of some changes in the network configuration.
- Connected TorBox to one of the available wireless networks in the TWM.
- Restarted TorBox --> TorBox automatically reconnected with the wireless network.