zotil/TorBox #89: How to connect to Tor from China

nyxnor, 2021-09-14:
Specific for China.

https://support.torproject.org/censorship/connecting-from-china/

### How to circumvent the Great Firewall and connect to Tor from China?
Users in China need to take a few steps to circumvent the Great Firewall and connect to the Tor network. First, get an updated version of Tor Browser: send an email to gettor@torproject.org with the subject "windows zh-cn" or other operating system (linux or macos)

After installing Tor Browser, you will probably not be able to connect directly to the Tor network, because the Great Firewall is blocking Tor. Therefore, the second step will be to obtain a bridge that works in China.

There are three options to unblock Tor in China:

* meek-azure: it looks like you are browsing a Microsoft website instead of using Tor. However, because it has a bandwidth limitation, this option will be quite slow. You can select meek-azure from Tor Browser's built-in bridges dropdown.
* Snowflake: uses ephemeral proxies to connect to the Tor network. It's available in Tor Browser stable version (Desktop and Android). You can select Snowflake from Tor Browser's built-in bridge dropdown.
* Private and unlisted obfs4 bridges: users will need to request a private bridge to frontdesk@torproject.org or, if they are tech-savvy, they can run their own obfs4 bridge from outside China. It's important to note that bridges distributed by BridgeDB (HTTPS, email), and built-in obfs4 bridges bundled in Tor Browser most likely won't work.

If one of these options below is not working, check your Tor logs and try another option.

radio24, 2021-09-22:
It seems that [a specific Telegram channel](https://t.me/tor_bridges) publishes OBFS4 bridges on a daily basis, which work in Iran. I don't recommend using these bridges, if not really necessary, because, without further investigation, it is not clear from where these bridges are coming. Nevertheless, it would be interesting to test if these bridges have a higher chance to work from China.

nyxnor, 2021-10-03:
Yeah, discovered that channel because of TorBox's torrc, but still skeptic of the method.

nyxnor, 2021-10-03:
This `Tor from China` could be a text on first use https://github.com/radio24/TorBox/issues/86 and/or a text on the [Bridges README](https://github.com/radio24/TorBox/blob/v.0.4.3/text/help-bridges-text). But mainly what I see is missing is `private` bridges part to explain that. The rest is complete.

radio24, 2021-10-03:
I'm not sure if this is not too much information. People don't like to read, and, in general, we have already long explanation texts. There is also the TorBox website and this GitHub page - if someone wants to know more, he will find additional information. Nevertheless, let's get specific, what would you like to add in which file?

nyxnor, 2021-10-03:
There are three options to unblock Tor in China:
* meek: Done!
* snowflake: Done!
* private: Private and unlisted obfs4 bridges: users will need to request a private bridge to frontdesk@torproject.org or, if they are tech-savvy, they can run their own obfs4 bridge from outside China. It's important to note that bridges distributed by BridgeDB (HTTPS, email), and built-in obfs4 bridges bundled in Tor Browser most likely won't work.

radio24, 2021-10-03:
Hmm... I don't think we should expand the text files on TorBox. We can add a link to this page or to the TorBox site. 

By the way: were you lately able to connect Snowflake? For me, the Snowflake server is always down and I cannot connect it.

nyxnor, 2021-10-03:
> Hmm... I don't think we should expand the text files on TorBox. We can add a link to this page or to the TorBox site.

Ok, site it is

> By the way: were you lately able to connect Snowflake? For me, the Snowflake server is always down and I cannot connect it.

Same for me, down.

radio24, 2021-10-03:
I will let the issue open as documentation.
Unfortunately, GitHub doesn't support the creation of a "Documentation" folder, so I decided to let all issues labelled with "Documentation" rest open.

nyxnor, 2021-10-03:
you can set wiki pages https://github.com/AnarchoTechNYC/CTF/wiki/Tor#localhost-bypass-attack

radio24, 2021-10-03:
Thank you for this suggestion. I already thought about that, but this will give me additional work, which I rather will spend on coding. At the moment, I'm stuck with some nasty wifi problems. That's why I couldn't start with the onion services yet.

closed 2021-11-11 by radio24