zotil/TorBox #94: Syntax improvement broke the code!

radio24, 2021-10-02:
Hey nyxnor
> * `! -z` for non zero or not null and `-z` for zero/null

There is a big problem: zero/null means `""` or `0`, but not `"0"`, because that is a string value. My test showed me that the following replacement in the pull requests broke the code:

**Old**
`if [ $CLEARNET_ONLY == 0 ]; then`

**New**
`if [ -z "$CLEARNET_ONLY" ]; then`

If `CLEARNET_ONLY=0` the "Old" will be true but new will be false. True would be the following:
`if [ -z $CLEARNET_ONLY ]; then`

Of course Shellcheck would moke because of the missing quotes.

radio24, 2021-10-02:
Half as bad! Only `torbox.lib` was affected at two positions - I will fix the code in some minutes.

closed 2021-10-02 by radio24

nyxnor, 2021-10-03:
Ok. Good it was solved.
I have the habit of using double quotes on variables, but this have to be done with attention. If the variable is quoted `"$var"`, the value of the variable needs to be quoted also `var="0"`. But I forgot that when doing those changes, sorry.