zotil/TorBox #99: changing hostapd wlan address

anvarit, 2021-10-15:
I'm using the torbox on a rpi 4. 
After installing everything works but the torbox uses the internal RPi WLAN adapter (wlan0)

When I try to modify the hostapd.conf file the change the adapter settings to wlan1 (Atheros AR9271) and reboot the rpi the setting is reverted back to wlan0

The internet connection is configured to ethernet and I just want to use the torbox as an access point but with the more powerfull AR9271

anvarit, 2021-10-17:
Understable to have a stable image that allows for tinkering and having a fallback method
Internet here is eth0. (POE hat)
My thoughts is that having an options that allows you to choose the wlan adapter would be the ideal solution. Ofcourse this can also bring new problems if you change wireless nics because they will become wlan2, 3 and so on. 

Now for me this isn't an issue, I reversed my setup. Used the wlan1 as internet input and put the eth0 in to a vlan and added a new SSID in my wireless infrastructure with that vlan. This solved my isse and gives me coverage all over the house. The wlan0 still broadcasts the Torbox ssid as mangement interface to the torbox.

anvarit, 2021-10-17:
Would be great, thanks for the great work that you already did!

closed 2021-10-17 by anvarit

radio24, 2023-07-11:
In TorBox v.0.5.3 in the Danger Zone submenu the failback which resets the AP to wlan0 during the boot can be disabled. --> Issue closed