zotil/TorBox #108: Include `/etc/tor/torrc.d`

nyxnor, 2021-11-09:
https://github.com/torproject/tor/blob/9d8b0c5bdc6f75891440dbe53e2572bf54e74aef/src/config/torrc.sample.in#L252

>Configuration options can be imported from files or folders using the %include
option with the value being a path. This path can have wildcards. Wildcards are 
expanded first, using lexical order. Then, for each matching file or folder, the following 
rules are followed: if the path is a file, the options from the file will be parsed as if 
they were written where the %include option is. If the path is a folder, all files on that 
folder will be parsed following lexical order. Files starting with a dot are ignored. Files 
on subfolders are ignored.
The %include option can be used recursively.
#%include /etc/torrc.d/*.conf

`%include /etc/tor/torrc.d/*.conf`

This can be added to the torrc or to `/usr/share/tor/tor-service-defaults-torrc`, which is what is done [here](https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/search?q=tor-hs-patch-config).

I think this can help manage hidden services, client bridges, server bridges, rest of configuration.

nyxnor, 2021-11-19:
It works when starting the tor process, but not when reloading for me at least. See [my issue](https://gitlab.torproject.org/tpo/core/tor/-/issues/40514#note_2761856).

### Plan

inside `/etc/tor`

`torrc`
```
%include /etc/tor/torrc.d
```

`torrc.d/main`
```
## DON'T CHANGE THE FOLLOWING 20 LINES!
######################################################
## Configuration for TorBox

Log notice file /var/log/tor/notices.log
VirtualAddrNetworkIPv4 10.192.0.0/10
AutomapHostsSuffixes .onion,.exit
AutomapHostsOnResolve 1
TransPort 127.0.0.1:9040
TransPort 192.168.42.1:9040
#TransPort 192.168.43.1:9040
DNSPort 127.0.0.1:9053
DNSPort 192.168.42.1:9053
#DNSPort 192.168.43.1:9053
SocksPort 127.0.0.1:9050
SocksPort 192.168.42.1:9050
#SocksPort 192.168.43.1:9050
SocksPort 127.0.0.1:9052 IsolateDestAddr
SocksPort 192.168.42.1:9052 IsolateDestAddr
#SocksPort 192.168.43.1:9052 IsolateDestAddr
ControlPort 127.0.0.1:9051
#ControlPort 192.168.42.1:9051
#ControlPort 192.168.43.1:9051
HashedControlPassword 16:E68F16640ED8C0F7601F5AA3D229D8DFD8715623CB055577F9434F7FB7
DisableDebuggerAttachment 0
```

`torrc.d/relay`
```
## THE CONFIGURATION OF THE BRIDGE RELAY STARTS HERE!
######################################################
## This will setup an obfs4 bridge relay.
#BridgeRelay 1
#ORPort 4235
#ExtORPort auto
#ServerTransportPlugin obfs4 exec /usr/bin/obfs4proxy
#ServerTransportListenAddr obfs4 0.0.0.0:443
#ContactInfo <address@email.com>
#Nickname TorBox042
#BridgeDistribution any
```

`torrc.d/bridges`
```
## TO OVERCOME CENSORSHIP, START HERE!
######################################################
## If you like to use bridges to overcome censorship, EDIT THE LINES BELOW!
## To use bridges, uncomment the three lines below...
#UseBridges 1
#UpdateBridgesFromAuthority 1
#ClientTransportPlugin meek_lite,obfs4 exec /usr/bin/obfs4proxy
#ClientTransportPlugin snowflake exec /usr/bin/snowflake-client -url https://snowflake-broker.torproject.net.global.prod.fastly.net/ -front cdn.sstatic.net -ice stun:stun.l.google.com:19302,stun:stun.voip.blackberry.com:3478,stun:stun.altar.com.pl:3478,stun:stun.antisip.com:3478,stun:stun.bluesip.net:3478,stun:stun.dus.net:3478,stun:stun.epygi.com:3478,stun:stun.sonetel.com:3478,stun:stun.sonetel.net:3478,stun:stun.stunprotocol.org:3478,stun:stun.uls.co.za:3478,stun:stun.voipgate.com:3478,stun:stun.voys.nl:3478

## Meek-Azure
#Bridge meek_lite 192.0.2.2:2 97700DFE9F483596DDA6264C4D7DF7641E1E39CE url=https://meek.azureedge.net/ front=ajax.aspnetcdn.com

## Snowflake
#Bridge snowflake 192.0.2.3:1 2B280B23E1107BB62ABFC40DDCC8824814F80A72

## OBFS4 bridges
##
## You have three ways to get new bridge-addresses:
## 1. Get them here https://bridges.torproject.org/
##    (chose "Advanced Options", "obfs4" and press "Get Bridges)
## 2. Or send an email to bridges@torproject.org, using an address
##    from Riseup or Gmail with "get transport obfs4" in the body of the mail.
## 3. (Not recommended, only if needed): Via Telegram: https://t.me/tor_bridges
#Bridge obfs4 144.217.20.138:80 FB70B257C162BF1038CA669D568D76F5B7F0BABB cert=vYIV5MgrghGQvZPIi1tJwnzorMgqgmlKaB77Y3Z9Q/v94wZBOAXkW+fdx4aSxLVnKO+xNw iat-mode=0
#Bridge obfs4 37.218.245.14:38224 D9A82D2F9C2F65A18407B1D2B764F130847F8B5D cert=bjRaMrr1BRiAW8IE9U5z27fQaYgOhX1UCmOpg2pFpoMvo6ZgQMzLsaTzzQNTlm7hNcb+Sg iat-mode=0
#Bridge obfs4 109.105.109.147:13764 BBB28DF0F201E706BE564EFE690FE9577DD8386D cert=KfMQN/tNMFdda61hMgpiMI7pbwU1T+wxjTulYnfw+4sgvG0zSH7N7fwT10BI8MUdAD7iJA iat-mode=2
```

`torrc.d/services`
```
HiddenServiceDir /var/lib/tor/services/ssh
HiddenServiceVersion 3
HiddenServicePort 22 127.0.0.1:22

HiddenServiceDir /var/lib/tor/services/xmpp
HiddenServiceVersion 3
HiddenServicePort 5222 127.0.0.1:5222
HiddenServicePort 5269 127.0.0.1:5269
```

radio24, 2021-11-19:
At the moment, for TorBox, I don't see any need to split the torrc up to several other files.

nyxnor, 2021-11-21:
can be closed.

closed 2021-11-21 by radio24