zotil/TorBox #114: TorBox and Pi-Hole

frenchfaso, 2021-11-13:
Hello there,
is it possible (and does it make sense) to run Pi-Hole alongside TorBox (perhaps as a docker container on the same raspberrypi)?
If yes, how should I configure the thing?
Thanks for TorBox!

radio24, 2021-11-13:
Hello frenchfaso
Recurrent, people ask about combining or implementing Pi-Hole with TorBox. This is why I will move this topic from the issue section to the discussion (FAQ). 

My personal opinion is that it makes not much sense to run Pi-Hole alongside TorBox, but finally, everyone has to decide for themselves. If a client connected to the TorBox performs a DNS request, TorBox will send this request through the tor network. TorBox has to „leak“ DNS requests to the Pi-Hole if you want to bring Pi-Hole into play. Additionally, Pi-Hole would have to be configured in that way that the DNS request will be routed through tor or at least that no third party (censor, spy, manipulated DNS server …) can see the DNS request or tie it to the client, who made the DNS request in the first place. In my opinion, if you want to block advertisements while browsing the web effectively, the best solution is to use a browser like [Brave](https://brave.com/) or [Tor Browser](https://www.torproject.org/download/) with the add-on [uBlock Origin](https://github.com/gorhill/uBlock/).

About implementing Pi-Hole into TorBox: [see here](https://www.torbox.ch/?page_id=112#could-you-incorporatepi-hole-to-remove-ads).

frenchfaso, 2021-11-13:
Thanks for the explanation!
Totally makes sense.
I ran a "pihole" search on the github issues, but I didn't think of looking through the faq on the website :sweat_smile:

closed 2021-11-13 by radio24