zotil/TorBox #116: How to limit Bandwith

dhuschde, 2021-11-16:
Hello, I currently use TorBox as my Guest Wi-Fi to secure my private one.
And now, as my Network isn't fast, I wanted to limit the Guest Wi-Fi to 5 MB/s. But I don't know how.

Do you have any Idea on how to do that directly in TorBox?
My Router sadly doesn't have that feature.

Thanks in advance

radio24, 2021-11-16:
Honestly, we didn’t study that option yet.

According to the [torrc manual](https://2019.www.torproject.org/docs/tor-manual.html.en), there are some settings, which you could try in the `torrc` file:

`BandwidthRate N bytes|KBytes|MBytes|GBytes|TBytes|KBits|MBits|GBits|TBits`

It would be interesting to hear from you if that solved your problem. In this case, we would think about including that option in one of the next TorBox versions.

dhuschde, 2021-11-16:
Seems to be working with this:
[image]

I now set it to 2 MB (according to [Tor Project](https://2019.www.torproject.org/docs/tor-manual.html.en), 250 KB would be 2...)

I tested it on fast.com. It isn't always under 2, but most likely.

As I am unsure, whether this is due to a bad connection to Tor or not, I will keep an eye on it.

dhuschde, 2021-11-17:
As my Router has a Traffic-Monitor, I kept an eye on this over the Router.
It seems to be working just fine for the Download speed. So you can safely add this to the Config menu.

And thank you very much for your Help.

radio24, 2021-11-17:
Glad to hear.
Thank you for the testing and the feedback.

radio24, 2025-08-03:
With the [Commit 80dfbc6](https://github.com/radio24/TorBox/commit/80dfbc6d1b1367a9c01ea3751152335b2d2730f8), the above-requested feature will be included in the upcoming TorBox v.0.5.5.

For testing it out, please use entry 5 in the Update and Maintenance sub-menu. In expert mode, paste the names of the developer branch ([TorBox-v.0.5.5](https://github.com/radio24/TorBox/compare/TorBox-v.0.5.5)) and perform the update.

Subsequently, add/change the following entries in the `run/torbox.run` file:
```
# SSH access from the Internet
# 0 - SSH access from the Internet is blocked (default on a TorBox / TorBox mini)
# 1 - SSH access from the Internet is allowed (default on a TorBox on a Cloud)
SSH_FROM_INTERNET=0

# SSH access from the Wifi
# 0 - SSH access from the WiFi is blocked
# 1 - SSH access from the WiFi is allowed (default)
SSH_FROM_WIFI=1

# SSH access from the Cable
# 0 - SSH access from the Cable is blocked
# 1 - SSH access from the Cable is allowed (default)
SSH_FROM_CABLE=1

# SSH access from the USB/VPN
# 0 - SSH access from the USB/VPN is blocked
# 1 - SSH access from the USB/VPN is allowed (default)
SSH_FROM_USB=1
```

Also, add/change the following entries in the `/etc/tor/rorrc` file:
```
#BandwidthRate 1 GB
#BandwidthBurst 1 GB
```

closed 2025-08-17 by radio24