zotil/TorBox #134: edit a temporary copy of the torrc instead of the directly editing it.

nyxnor, 2021-12-24:
Benefits:
* not editing the main configuration directly is safer
* verifying its config before running is safer
* restarting/reloading the configuration only after verifying it is ok is safer to now make the tor daemon fail

To get a hold of what it does, see:
* [function safe_edit](https://github.com/nyxnor/onionjuggler/blob/d6be2cb3bb41d8c50fb6e79eaae0d6923ef216a8/onionjuggler-cli#L272), will create a temporary copy of any file inside the temporary directory if writable
* [function verify_config_tor](https://github.com/nyxnor/onionjuggler/blob/d6be2cb3bb41d8c50fb6e79eaae0d6923ef216a8/onionjuggler-cli#L307), only show warns and if wrong, exit without modifying the default torrc
* [function signal_tor](https://github.com/nyxnor/onionjuggler/blob/d6be2cb3bb41d8c50fb6e79eaae0d6923ef216a8/onionjuggler-cli#L335) the calls verify_config_tor and if ok, procede to signal tor to reload/restart.
* [usage of the temp file](https://github.com/nyxnor/onionjuggler/blob/d6be2cb3bb41d8c50fb6e79eaae0d6923ef216a8/onionjuggler-cli#L530) that will save its name as a variable and use it to reference on the script.

It also don't put the temp file back in place if there is no difference, using the `cmp -s` command. But I still signal tor either way because it is not only the torrc that could be changed, but also the files inside the tor data dir.

I think this can be done to TorBox when editing the torrc. It is like a [`vidoas`](https://github.com/slicer69/doas/blob/master/vidoas)/[`visudo`](https://www.sudo.ws/docs/man/visudo.man/) but for the torrc through the advanced configuration editor.

> visudo edits the sudoers file in a safe fashion, analogous to vipw(8). visudo locks the sudoers file against multiple simultaneous edits, performs basic validity checks, and checks for syntax errors before installing the edited file. If the sudoers file is currently being edited you will receive a message to try again later.

So the plan is the following:
* Implement the above functions for every modification of the torrc through scripts, this will take some time but it is worth it.
* Implement `visudotor` to save a temp file, modify it with the EDITOR, verify it, if ok save to default location and signal tor. It is almost the same as the above but includes the editor part.

nyxnor, 2021-12-25:
I created `vitor` now: https://github.com/nyxnor/onionjuggler/blob/main/bin/vitor, is is very tiny, less than 100 lines of code so there should be no problems. Also, it is portable so you should just need to save it to PATH (/usr/local/bin for example and all good).
It should be run as root using `sudo` or `doas` and you if you don't specify the file, it fallback tor /etc/tor/torrc. The tor_user must be specified (debian-tor for debian systems) if the `User` option is not contained in that configuration.
Vitor is just the editor part, the post above is still valid for other scripts that don't use the editor but sed the torrc for example.

radio24, 2022-01-01:
Wow, great tool!
I implemented it in the local copy of TorBox v.0.5.0 and it will be commited when I'm finished with the TFS integration.

nyxnor, 2022-01-22:
I changed vitor repo to be on its own as it does not have entirely to do with OnionJuggler and HiddenServices but with torrc on a whole.

https://github.com/nyxnor/vitor

Also made a lot of improvemens and fixes. Log messages are more helpful.

closed 2022-07-13 by radio24