zotil/TorBox #137: Replace wpa2 with Wpa3-SAE?

radio24, 2022-01-02:
### Discussed in https://github.com/radio24/TorBox/discussions/130

<div type='discussions-op-text'>

<sup>Originally posted by **ipwebnl** December  2, 2021</sup>
Wpa2 to easy to crack with simple gpu. And it is very easy to decrypt WPA2-psk traffic. 

Love the project but think that this is the biggest weakness.</div>

Jongjan88, 2023-05-07:
💪💪❤️❤️

radio24, 2025-08-03:
In preparation for the next TorBox version, I looked into this issue again and tried to create a WPA3-SAE Access Point (AP).

For a WPA3-SAE AP under Linux using `hostapd`, a few prerequisites are necessary:
- WPA3-SAE is only supported from `hostapd` 2.7 or newer and has to be compiled with SAE support.
- AP side: Compatible WLAN hardware and kernel drivers (e.g. `nl80211`) are needed.
- Client side: The clients that are to connect must support WPA3-SAE.

Regarding the configuration, `hostapd` has to be added with the following lines:
- `wpa_key_mgmt=SAE` for WPA3-SAE only or `wpa_key_mgmt=SAE WPA-PSK` for WPA2/WPA3 mixed mode.
- `ieee80211w=2` for WPA3-SAE only or `ieee80211w=1` for WPA2/WPA3 mixed mode.

The `hostapd` package from the official Raspberry Pi OS repositories is up to date and supports WPA3-SAE. However, clients cannot connect to the AP using the onboard Wi-Fi on most Raspberry Pi models (3, 4, 5). The issue appears to be related to the incorrect password, but unfortunately, the problem seems to lie in the [chipset of the onboard Broadcom-Chip](https://docs.raspap.com/ap-basics/)/firmware/driver. This situation is independent of the Raspberry Pi model, the firmware and the Linux kernel.

Also, all distributions seem to face identical limitations because they all use:
- same hardware constraints
- same Broadcom/Cypress firmware blobs
- same brcmfmac kernel driver

**Possible workaround**
WPA3 support for an AP may be possible with some WiFi USB-adapters. However, we had a very mixed experience using WiFi USB-adapters for AP, even with WPA2. Therefore, we believe this solution is only suitable for experts and shouldn't be integrated into the TorBox menus at this time.