zotil/TorBox #138: Problem: improve curl proxy method

radio24, 2022-01-03:
@nyxnor I may found a problem with the following use of `curl`, but maybe I do something wrong:

`OCHECK=$(curl -x socks5h://127.0.0.1:9050 -m 6 -s $CHECK_URL)` used [here](https://github.com/radio24/TorBox/blob/cff481d1071c34a480abeab80a521dd2d2e49d22/lib/torbox.lib#L172) gives a `$?` of `97` .

nyxnor, 2022-01-03:
I haven't seen the exit code 97 yet neither on the man page https://curl.se/docs/manpage.html#exit-codes

```sh
$ curl -x socks5h://127.0.0.1:9050 -m 6 -s https://check.torproject.org/api/ip
{"IsTor":true,"IP":"22.333.444.555"}
$ echo $?
0
$ curl -x socks5h://127.0.0.1:9050 -m 6 -s http://google.com
<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
$ echo $?
0
```

nyxnor, 2022-01-03:
First try was correct domain, second try is incorrect and shows 97
```sh
$ OCHECK=$(curl -x socks5h://127.0.0.1:9050 -m 6 -s http://google.com)
$ echo $?
0
$ OCHECK=$(curl -x socks5h://127.0.0.1:9050 -m 6 -s http://google.coa)
$ echo $?
97
```

radio24, 2022-01-03:
Ok, found the problem - it was the URL. Thanks for helping!

closed 2022-01-03 by radio24