zotil/TorBox #143: TorBox + VPN

biscwii, 2022-01-20:
Hi !

Thanks for the great work done with TorBox !
I have set up TorBox on a RPI3 and my devices use tor through the wifi provided by TorBox. But obviously it will work only if connect my device to the wifi network, so it will work only at home. Is it possible to set up a vpn server on the same RPI3 to connect to it from outside and be redirected to tor by TorBox? So my question is : Is it possible to replace the wifi connection by a VPN tunnel to connect to TorBox?

Thanks !

Zotil, 2022-01-20:
It could be possible setting up a VPN service inside the TorBox, connecting it through Tor socket.
If "outside" means out of home: expose the VPN service's port to public ip address.

You don't need to replace wifi for this.

TorBox comes with OpenVPN installed, so you can configure the service without installing any additional

biscwii, 2022-01-23:
I tried to set up the vpn server using pivpn (as it is really easy to use) with OpenVPN. When I reboot, TorBox main menu tells me vpn is enabled. But I can't connect a client to it. I heard that I have to use TCP because UDP won't work with tor, but it still doesn't work. Do I need to change something in iptables?
(Thanks for helping me with this, I'm sure that giving the ability to TorBox to connect all your device to it even when your are not at home could really improve its usage)

radio24, 2022-01-23:
Hey,
So far, TorBox doesn't support client data stream coming from a vpn connection. To set it up with pivpn is not enough because iptables has to configured accordingly. Also, torrc has to be configured with the right IP address (or tun0 has to have the right IP address to work with tor) - see [here](https://github.com/radio24/TorBox/blob/2b27450e2ca3bad5fc50d4e028c41b46fd76078b/menu#L203). Most probably that have to be a separate IP address space because it should work together with wifi and cable clients.

In your case, TorBox will falsely tell you that vpn is enabled because in [check_interface_with_internet()](https://github.com/radio24/TorBox/blob/2b27450e2ca3bad5fc50d4e028c41b46fd76078b/lib/torbox.lib#L71) we make only a simple check if `tun0` exists and think that is a Internet connection.

You are also right that UDP will not work with tor. In any case VPNs has to be configured to use TCP.

We have the request to implement a VPN server on our to-do list, but at the moment we have other priorities (alternative pass through captive portal which could lead to randomize MAC addresses and also to a tighter torrified TorBox; further implementation of Onion Services).

radio24, 2024-01-26:
Test VPN server and TorBox on a cloud capabilities and contribute with feedback: https://www.torbox.ch/?p=3522

radio24, 2024-12-30:
With TorBox v.0.5.4, we added the possibility to install an OpenVPN server, which you can connect from a client after creating an .ovpn file on the TorBox and adding it to the [OpenVPN Connect](https://openvpn.net/client/) on your client. The corresponding menu entry is in the configuration sub-menu (entry 20). The documentary on the website isn't up-to-date yet, but the information in the menu entry should be self-explanatory.

closed 2024-12-30 by radio24